rusefi/rusefi #1590: Build Orthogonality Issues

chuckwagoncomputing, 2020-07-11:
If any gen_config fails, all firmware builds fail.
If console build fails, all firmware builds fail.
If simulator build fails, all firmware builds fail.
And probably others too.

chuckwagoncomputing, 2020-07-11:
Slack transcript:
```
Matthew Kennedy 3:23 PM
other than the fact that PR gates should each be relatively orthogonal - right now a console failure is causing all of the firmware builds to fail

dbh97 - chuckwagoncomputing  4:41 PM
Why? It shouldn't.

Matthew Kennedy  4:41 PM
https://github.com/rusefi/rusefi/pull/1582
have a look at the builds

dbh97 - chuckwagoncomputing  4:44 PM
Oh right, I thought you meant that if it failed on one build it would fail all of them.
The console build is required because it is included in the bundles, which is the main point of the build (in addition to ensuring firmware builds correctly). This is how the Jenkins job build_extra_bundles has always been.

Matthew Kennedy  4:45 PM
the trouble is that it hurts the SnR on PR gate builds
Ideally one problem causes one failure
and the builds are orthogonal

dbh97 - chuckwagoncomputing  4:47 PM
So how should bundles be built if the console build fails?

Matthew Kennedy  4:47 PM
I'm saying that we don't need to build bundles for PR
the current behavior is great for master - we want to actually emit the thing (or not if anything is broken)
but on a PR, a failure of the firmware build when a console unit test fails is difficult to read through
the only product of the PR gate should be "did X build ok? yes/no" for each X we care about

dbh97 - chuckwagoncomputing  4:53 PM
10-4
I'm not sure if you've been in on this discussion, but @Andrey is concerned about keeping Jenkins working, and in order to keep that goal, I've been setting things up to use the same scripts for Jenkins and Github Actions.

Matthew Kennedy  4:55 PM
by build I mean "build and test", so it's fine to be running the console test that's failing, but that failure should just be confined to the "console build/test" action
yes, that's a reasonable goal
easy enough to have a script for each, one that builds for the intent of bundle, and one that builds just for PR

dbh97 - chuckwagoncomputing  4:58 PM
I'll think this through some more.
```

chuckwagoncomputing, 2020-07-11:
Also see #1587, there is some discussion there.

rusefi, 2020-07-11:
But what's the proposed behavior? How do we produce a bundle which requires a console without console building?

One way to handle this would be chained jobs with artifacts passing, i.e. a pipeline:

1) gen config produces gen config stuff and pushes thus modifying state of git
2) console build builds console

x) only if #1 and #2 succeed we can start building bundles while somehow consuming console artifact from #2 (#1 artifact would be in git but we would have to take fresh git state not the commit which caused the pipeline to trigger)

I am not convinces that this is worth the efforts. it's definitely nice to have but things work as as and I am happy to accept the status quo.

chuckwagoncomputing, 2020-07-11:
Matt's suggestion in the slack discussion was to only produce a bundle on push to master.

chuckwagoncomputing, 2020-07-11:
```
 +-------------+-----------+Jenkins
 |             |               +
 |             |         +-----+-------+
 |             |         |for each brd |
 |             |         +-----+-------+--------+-----------+
 |             |               |                |           |
 v             v               v                v           v
Sim         Console        config_board      firmware     package
 ^             ^               ^   ^          ^ ^            ^
 |             |               |   |          | |            |
 |             |               |   +------+---+ |            |
 |             |               |          |     |            |
 |             |               |          |     |            |
 +-----------------------------+----------------+------------+
               |                          |
 +----------------------------+       +---+---------+--------------+
 |for each brd | in workflow  |       |for each brd | in workflow  |
 +-------------+--------------+       +-------------+--------------+
                Push                          PR

```

commit 2aac840c (2020-07-14): Fix #1590 (#1602)

* Add checkout

* remove empty with statement

* Actually we do need push depth

* convert gen_upload_docs.bat

* convert run.bat

* forgot to add run.sh

* convert gen_upload_docs.bat

* remove checking from bats

* change config

* fix path

* fix path again

* fix file extension

* remove escapes

* added missing quotation and escapes

* convert build_current_bundle

* convert clean_bootloader

* convert compile_bootloader_discovery

* convert compile_bootloader

* add env var to skip upload

* remove checking from compile.bat

* working on push.yaml

* finish sorting

*  [... cut]
closed 2020-07-14 by rusefi