rusefi/rusefi #1592: Injectors #1 and #5 stick open

mck1117, 2020-07-11:
[image]

Something is wrong with the overlapping injection logic with fuel scheduling.  The counter is getting incremented for those cylinders, but never decremented.  Commenting out the overlapping injection logic makes it go away.

mck1117, 2020-07-11:
[TunerStudioDebugPackage.zip](https://github.com/rusefi/rusefi/files/4906455/TunerStudioDebugPackage.zip)

mck1117, 2020-07-11:
Debugging notes:
during schedule rebuild in the transition from batch to sequential, we cancel all outstanding injector open/close events, and on my engine this process occurs during the time that cylinders 1 and 5 have an injector open.

rusefi, 2020-07-15:
@mck1117 can you please elaborate on how was this screenshot produces? real vehicle cranking? some sort of simulation?

can you please attach raw .logicdata?

did you try reproducing this with a unit test?

mck1117, 2020-07-15:
Here's the `.logicdata`:
[bad-batch-to-seq.zip](https://github.com/rusefi/rusefi/files/4923287/bad-batch-to-seq.zip)

This was first observed on a real engine that actually stuck the injectors open, dumping 350cc/min of gas in to cylinders 1 and 5 😲

rusefi, 2020-07-15:
how was the logic data captured? looks like capture has started before the event.

mck1117, 2020-07-18:
of course, I started the capture, then activated self stim

rusefi, 2020-07-18:
@mck1117 what would I need to do to reproduce this in NJ?

mck1117, 2020-07-18:
This happens when a few conditions are met:
1. During cranking, the sync point where we switch mode from cranking -> running occurs while one or more injectors are open
2. Before the injector closes, another trigger tooth (which could be the same tooth from 1) triggers a reschedule of the open injector
  
This resets and re-schedules any existing events that were already scheduled - including the one that was supposed to decrement `overlappingCounter` and close the injector.  This causes `overlappingCounter` to get out of sync with reality, as indeed the injector has been opened once that had no matching closure.

The fix is to reset that overlap counter any time we do a full reschedule that has the potential to get the world out of sync.

mck1117, 2020-07-18:
Here's a tune that triggers the issue:
[TunerStudioDebugPackage.zip](https://github.com/rusefi/rusefi/files/4942626/TunerStudioDebugPackage.zip)
And corresponding .logicdata when enabling self stim with that tune:
[new_bad.zip](https://github.com/rusefi/rusefi/files/4942642/new_bad.zip)
[image]

rusefillc, 2020-07-20:
logicdata work in progress

[image]

rusefillc, 2020-07-20:
[image]

PR #1606 (2020-07-21): reset overlap counter on fuel reschedule
fixes #1592
closed 2020-07-21 by rusefillc