rusefi/rusefi #1594: Cranking -> Running transition is a mess

mck1117, 2020-07-11:
Cranking and running modes are both sequential, and this mess happens during the transition from cranking to running:

[image]

mck1117, 2020-07-11:
Note that the duration of these two intervals is approximately the same:
[image]

What's happening is that the first small injection is scheduled, then the injection phase is updated, then at the next tooth its scheduled AGAIN.  Since we target the end-of-injection timing, the first injection got scheduled expecting the duration of the longer cranking pulse, not the shorter running pulse.

rusefi, 2020-07-15:
Poke @mck1117

rusefillc, 2020-09-06:
@mck1117 what is the state of this issue?

mck1117, 2021-09-29:
there are more doubled injections here: https://github.com/rusefi/rusefi/issues/3275#issuecomment-927444156

rusefillc, 2022-05-26:
@mck1117 poke :)

rusefillc, 2022-09-01:
@mck1117 I assume you were busy for the last five months?

Closing since no one cares to use many words, not actionable as is.

mck1117, 2022-09-01:
> no one cares to use many words

huh?

mck1117, 2022-09-01:
This is still definitely a bug.

mck1117, 2022-09-01:
It happens roughly like this:

1. Long injection scheduled, so we start the injection very early to hit the requested end-of-injection time.
2. Injection duration becomes very short, so it starts when the long injection would've, but then completes very early.
3. New start time based on the short injection hasn't happened yet, so we start it again, getting a doubled injection.

rusefillc, 2022-09-01:
Related to #4507