rusefi/rusefi #1597: [TCU] manual shift buttons

rusefi, 2020-07-11:
as a preparation for TCU:

we already have
tcuUpshiftButtonPin
tcuDownshiftButtonPin

in configuration - and that's all we have.

1) we need to add TS menu to show these two controls

2) we need the code to count currently desired gear, with ringing prevention - i.e. with a min duration between toggle events. we probably should not bother with event capture, just polling button state from slow loop should be fine. See startStopButtonPin implementation

3) we need TS gauge showing currently desired gear

rusefi, 2020-07-11:
@chuckwagoncomputing is this something you would be interested to implement?

chuckwagoncomputing, 2020-07-11:
Sure, I can do that.

rusefillc, 2020-09-03:
#1750 is relavant

rusefillc, 2020-09-03:
close or not close yet?

chuckwagoncomputing, 2020-09-03:
Done