rusefi/rusefi #1604: Allow for negative TPS values for error detection and enrichment

stefanst, 2020-07-13:
It appears that currently the displayed TPS value is constrained to >=0. This is *apparently* also true for some calculations.
However, according to folklore, sometimes the actual value, even if negative, is used for calculations or gating. I have no proof of this, it was mentioned on slack. 
In addition, for TPS enrichment, small movements at near 0% throttle are important. If there is drift and the throttle position that used to be 0% is now -2% for some reason, we lose those critical first 2% for enrichment, if the value is clamped at 0%.

Are there any downsides to allowing negative values?

OrchardPerformance, 2020-07-30:
I am not sure this is the case, I have had my TPS values reading negative due to damaged brushes in the TPS. 

Also it has not caused and real downsides with the TPS reading negative when closed, apart from the obvious lag in reaching the idle threshold and fuel cutoff threshold. Made things a bit jerky was all.

mck1117, 2020-08-21:
We do allow negative TPS values.  In fact, we allow values as far out as those specified by the first two parameters on the TPS settings page.

OrchardPerformance, 2021-03-06:
@rusefillc - This can be closed as we DO allow negative TPS values based on the user defined limits.

closed 2021-03-28 by mck1117