rusefi/rusefi #1637: DC motor idle valve not ETB

rusefillc, 2020-07-25:
@Abricosvw has https://en.wikipedia.org/wiki/Volkswagen_Passat_(B4) which has cable driven throttle and DC motor based IAC

We would love to support this hardware configuration

According to @Abricosvw there are Toyotas the the same setup

See https://www.youtube.com/watch?v=dR5kFb1eiqw

rusefillc, 2020-07-25:
@mck1117 any interest?

Abricosvw, 2020-07-25:
[image]
[image]

Abricosvw, 2020-07-25:
Toyota have 2 flaps
One cable
One Etb idle

Abricosvw, 2020-07-25:
[image]
[image]
[image]

Abricosvw, 2020-07-25:
Nissan have 2 flaps
One cable
One Etb idle

rusefillc, 2020-07-25:
We shall focus on VAG only here to keep things easier

rusefillc, 2020-07-26:
https://youtu.be/fQSfJgjDjwU

mck1117, 2020-07-26:
All of the "second throttle plate" options are traction control, IIRC.  They're sprung open, and closed by the traction control computer to manage wheel slip.  I think those engines have a normal idle solenoid.

rusefillc, 2020-07-26:
[image]

rusefillc, 2020-09-06:
@Abricosvw reports that this works now

commit 0bbe159e (2020-07-26): Weird VAG: DC motor idle valve not ETB fix #1637
closed 2020-09-06 by rusefillc