rusefi/rusefi #1638: bootloader not well?

rusefillc, 2020-07-25:
We have up to three issues around bootloader compilation

1) looks like it does not compile
2) GHA does not notice or just uses different compiler and it works on GHA but not on Windows Jenkins (this has happened before)
3) Windows jenkins does not fail on bootloader failure?!

@chuckwagoncomputing @mck1117 can you please address at least some of these issues?

```Compiling persistent_store.cpp
Compiling io_pins.cpp
Compiling efilib.cpp
Compiling pin_repository.cpp
Compiling rusefi_stubs.cpp
Compiling dfu.cpp
Compiling main.cpp
Linking blbuild/bootloader.elf
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: blbuild/obj/efi_gpio.o: in function `OutputPin::setValue(int)':
efi_gpio.cpp:(.text._ZN9OutputPin8setValueEi+0x8): undefined reference to `perfEventBegin(PE)'
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: efi_gpio.cpp:(.text._ZN9OutputPin8setValueEi+0x42): undefined reference to `perfEventEnd(PE)'
collect2.exe: error: ld returned 1 exit status
make: *** [../ChibiOS/os/common/startup/ARMCMx/compilers/GCC/rules.mk:262: blbuild/bootloader.elf] Error 1
make compilation failed
/cygdrive/c/Program Files (x86)/Jenkins/workspace/integration_and_primary_bundle/firmware/bootloader
Sat 07/25/20 15:31:52.52
build_current_bundle: will be Erasing chip
build_current_bundle: Erasing chip
Entering flash_erase407.sh
/cygdrive/c/Program Files (x86)/Jenkins/workspace/integration_and_primary_bundle/firmware
Invoking openocd...
Open On-Chip Debugger 0.10.0 (2019-07-15) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
```

rusefillc, 2020-07-25:
@mck1117 @chuckwagoncomputing at least it fails now

```
Compiling persistent_store.cpp
Compiling io_pins.cpp
Compiling efilib.cpp
Compiling pin_repository.cpp
Compiling rusefi_stubs.cpp
Compiling dfu.cpp
Compiling main.cpp
Linking blbuild/bootloader.elf
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: blbuild/obj/efi_gpio.o: in function `OutputPin::setValue(int)':
efi_gpio.cpp:(.text._ZN9OutputPin8setValueEi+0x8): undefined reference to `perfEventBegin(PE)'
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: efi_gpio.cpp:(.text._ZN9OutputPin8setValueEi+0x42): undefined reference to `perfEventEnd(PE)'
collect2.exe: error: ld returned 1 exit status
make: *** [../ChibiOS/os/common/startup/ARMCMx/compilers/GCC/rules.mk:262: blbuild/bootloader.elf] Error 1
ERROR: bootloader compilation failed
FAILED TO COMPILE BOOTLOADER
build ERROR DETECTED
```

mck1117, 2021-03-08:
we can close this? bootloader builds fine these days and is tested by CI

commit a81a5e3c (2020-07-25): bootloader not well? fix #1638
closed 2021-03-08 by rusefillc