rusefi/rusefi #1649: gerber comparison toolset

rusefillc, 2020-07-29:
Problem statement:
we make PCB changes and we are not good documenting PCB changes
we need some way to see "diff" between two PCB revisions

proposal:
scripts? to compare gerber archieves

maybe there is something ready? maybe there is a command line gerber to image? maybe there is a command line image comparison similar to https://online-image-comparison.com/

mck1117, 2020-07-29:
https://github.com/Gasman2014/KiCad-Diff is worth investigating

chuckwagoncomputing, 2020-07-29:
https://github.com/jnavila/plotkicadsch - for SCH

rusefillc, 2020-08-14:
@chuckwagoncomputing "diffs" created https://rusefi.com/docs/diffs/

what else? can you please write up what do we have? maybe write it up as a PR with comments on top of the file?

right now I have no clue what does this action diff :) while we have a lot of hardware here in rusefi/hardware folder - the real drama happens in external repositories like https://github.com/rusefi/hw_microRusEfi

the most reused repo at the moment is https://github.com/rusefi/kicad-libraries that one is a sub-module for many repos including both MRE and rusEFI

chuckwagoncomputing, 2020-08-14:
Oh I didn't realize that most of the hardware wasn't here. I'll have to duplicate this script on those repos.

rusefillc, 2020-08-14:
https://github.com/rusefi/kicad-libraries/issues/17 created to compare between two releases

actually for automation a diff between two consecutive commits is only half of ideal toolset.

i think we also need a diff between master and previous release, that case with 5 changes to master we still compare with previous release in addition to five separate changes

which means that we need a clear protocol on how to identify previous release. we can have release_version.txt file with tag name or branch name maybe?

chuckwagoncomputing, 2020-08-15:
Are there any boards in this repo that need release version diff tracking?

rusefillc, 2020-08-15:
I've created version.txt at https://github.com/rusefi/rusefi/tree/master/hardware/Breakout_134pin_7-967288-1-Connector

next step @jharvey making changes to that board and we would be able to test the magic