rusefi/rusefi #1653: TS SD integration

rusefillc, 2020-07-29:
TS seems to have some SD download functionality, looks like binary format - maybe yet another binary format?

[image]

rusefillc, 2020-07-29:
```
      menu = "Data Logging"
        subMenu = sdcard_datalog, "SD Card Datalogging"
        subMenu = std_ms3SdConsole, "Browse / Import SD Card",0, {log_style_on2 && (tsLocalCanId == tsCanId) }

```

rusefillc, 2020-07-29:
```
   indicator = { sd_status & 1}, "No SD", "SD in",             white, black, green, black
   indicator = { sd_status & 4}, "SD ready", "SD ready",       white, black, green, black
   indicator = { sd_status & 8}, "SD Log", "SD Log",           white, black, green, black
   indicator = { sd_status & 16}, "SD Err", "SD Err",           white, black, red, black
```

rusefillc, 2020-07-29:
```
    sd_filenum = sd_filenum, "SDcard file number", "", 0, 9999, 9999, 9999, 9999, 9999, 0 ,0
    sd_error = sd_error, "SDcard error", "", 0, 255, 255, 255, 255, 255, 0 ,0
    sd_status = sd_status, "SDcard status", "", 0, 255, 255, 255, 255, 255, 0 ,0
    sd_phase = sd_phase, "SDcard phase", "", 0, 255, 255, 255, 255, 255, 0 ,0

```

```
   sd_status        = scalar, U08,   70,  "", 1.0, 0.0
   sd_filenum      = scalar, U16,  346, "", 1, 0
   sd_error        = scalar, U08,  348, "", 1, 0
   sd_phase        = scalar, U08,  349, "", 1, 0

```

rusefillc, 2020-07-29:
[image]

rusefillc, 2020-07-29:
http://www.msextra.com/doc/pdf/Megasquirt_Serial_Protocol-2014-10-28.pdf

[Megasquirt_Serial_Protocol-2014-10-28.pdf](https://github.com/rusefi/rusefi/files/4996186/Megasquirt_Serial_Protocol-2014-10-28.pdf)

rusefillc, 2020-08-01:
```
I 200731 204600.064 [BinaryProtocolServer2] BinaryProtocolServer - Got command command r/114
I 200731 204602.882 [BinaryProtocolServer2] BinaryProtocolServer - Got r [114, 0, 7, 2, 77, 0, 8]

rusefillc, 2020-08-01:
```
I 200731 212221.348 [BinaryProtocolServer2] BinaryProtocolServer - Got command command w/119
I 200731 212224.431 [BinaryProtocolServer2] BinaryProtocolServer - Got w [119, 0, 17, 0, 0, 0, 1, 4]

rusefillc, 2020-08-01:
```
I 200731 213511.233 [BinaryProtocolServer3] BinaryProtocolServer - Got command SD_R_COMMAND
I 200731 213518.766 [[proxy] TCP reader2] IoStream - IncomingDataBuffer: 13 byte(s) arrived
I 200731 213518.770 [BinaryProtocolServer3] BinaryProtocolServer - Got r [114, 0, 17, 0, 0, 0, 16]

rusefillc, 2020-08-01:
[log.txt](https://github.com/rusefi/rusefi/files/5010691/log.txt)

rusefillc, 2020-08-01:
@andreika-git http://www.fileformat.info/format/compress/compress.magic

rusefillc, 2020-08-01:
Progress

[image]

rusefillc, 2020-08-05:
SD would not delete files with shorter than 4 symbols file name

[image]

rusefillc, 2020-08-06:
@andreika-git can you guess date encoding?

sample data

```
68 65 6C 6C 6F 31 32 33 6D 6C 71 01 00 00 00 00 10 11     12 13 14 15     date > 16 17 18 19 1A 1B <  A1 3A 00 00 
68 65 20 20 20 20 20 20 6D 6C 71 01 00 00 00 00 1A 1B     1C 1D 1E 1F     date > 20 21 19 1A 1B 1C <  00 04 00 00 
5F 33 33 33 6F 31 32 33 6D 6C 71 01 00 00 00 00 24 25     26 27 28 29     date > 2A 2B 1A 1B 1C 1D <  40 42 0F 00 
```

UI
[image]

rusefillc, 2020-08-06:
[image]

[image]

andreika-git, 2020-08-07:
> @andreika-git can you guess date encoding?
Yes, it's a standard DOS/VFAT datetime format.

rusefillc, 2020-08-08:
We are 90% compatible with just Tuner Studio but only 90% :( what's killing our compatibility is TS requirement for gzip compression. I will email Phil asking if he had drop that requirement but for now we have to use our plugin not TS.

[image]

rusefillc, 2020-08-08:
Closing this ticket with initial implementation, minor follow-up activity #1681

closed 2020-08-08 by rusefillc