rusefi/rusefi #1658: Bug: Trigger gap-check toothDuration array samples time at any event.

960, 2020-07-30:
There are a bug in trigger_decoder.cpp, so the gap-check doesnt do whats intended.

The array toothDurations[i] should only capture one edge, but instead it capture each event.
[image]
[image]

This means we get a array that looks like this at the gap:

With a VR-sensor, this means the tooth-durations will vary with rpm, and return errors.
(Falling edge in gap)
 
We also get very few actual teeth for the gap-check, so it works by luck.

216148, 431631, 432107, 216051, 215988

```
#define getCurrentGapDuration(nowNt) \
	(isFirstEvent ? 0 : (nowNt) - toothed_previous_time)

	efitick_t currentDurationLong = getCurrentGapDuration(nowNt);

	toothDurations[0] = currentDurationLong > 10 * NT_PER_SECOND ? 10 * NT_PER_SECOND : currentDurationLong;

for (int i = GAP_TRACKING_LENGTH; i > 0; i--) {
			toothDurations[i] = toothDurations[i - 1];
		}
toothed_previous_time = nowNt;
```

rusefillc, 2020-07-30:
Not an issue, code comments improved to reduce the chance of confusion.

closed 2020-07-30 by rusefillc

960, 2020-07-30:
Not a issue?

toothed_previous_time are sampled at any event, and fills the array with both tooths and gaps.

I changed this to only sample at one edge, and see only the toothwidths + one long gap for missing tooth.
I also added correction for +-1 tooth.
That took care of my trigger issues.

Can you explain how/why it should be like it is?
Just wondering..

rusefillc, 2020-07-30:
I believe that "toothed_previous_time are sampled at any event" is a FALSE statement

anything after the "toothed_previous_time are sampled at any event" FALSE statement is, well, not an issue.

See https://github.com/rusefi/rusefi/commit/74f8efd7e15778b511458297380c4ae99279711f commit which improved source code comments

960, 2020-07-31:
Ok, do however you want it.. 
I debugged while driving, and the strange tooth-times are caused by this edge moving a lot with VR.

Making one wide, and one narrow tooth..

Thats the two higher values you see in array

[image]

mck1117, 2020-07-31:
That's actually not your problem.  The problem is that these edges got missed:
[image]

The way that we count teeth has nothing to do with interrupts getting dropped, which is already noted at #1488.

For us to help you, you need to:
1. reproduce the issue using UNMODIFIED firmware & TS ini
2. share that msq

960, 2020-07-31:
Yeah, that was just a random image I drew at.

The case is the car doesnt stop anymore, so I dont notice the scheduler-thing.
It still trips the other codes, 6045 etc.. but after changing the gap-checking, at least it doesnt stop :-)

mck1117, 2020-07-31:
Ok, so is there an issue or not? Does it reproduce on master or not?

960, 2020-07-31:
It's the same whatever firmware you use, just start debugging and see what values that goes in there.

Drama Queen spent only seconds to conclude this is not a issue and closed it.

I would call it a issue.
I think the correct durations to fill this array with are those shown in the bosch-screenshot.

These are the ones that ends up there

[image]

I changed it to capture times at only one edge(in trigger_central.cpp), and used these booleans instead.

[image]

rusefillc, 2020-07-31:
insults are no acceptable, let me block you here.

960, 2020-07-31:
I would call it a insult to close this in seconds, and claim everything are right without even checking.

This is the same friggin story every time.

960, 2020-07-31:
Also, it syncs when toothDurations[0] are larger. 
This is wrong as well..

[image]

960, 2020-07-31:
This is how the array looks in my code, so I dont need any help with this at all.
I tried to post a bug , that took you two seconds to rename to not beeing a bug and close.

I dont care about you at all Andrey, but I care about others that might get stuck or blow their engine over this.

Even if I think you are a sick creep, I would still like to see the project succeed.
When you go ahead with your closed-source plans maybe matt or someone else will take over and make it happen.

[image]

mck1117, 2020-07-31:
Running code from master, I see tooth durations that are only looking at one edge, so the gaps are in the ratio 1-1-1-3-1 (etc).  What is the EXACT configuration you were using in your first post where you saw 1-2-2-1-1-1?

mck1117, 2020-07-31:
Either way, whatever you did is just masking the real issue of missing teeth as identified in #1488.  That's still an issue, since we schedule all injection and ignition events off of trigger teeth.  If a tooth gets missed, everything for the rest of the cycle will end up late by the number of teeth missed (for example, 6 degrees for a 60-2 and one tooth lost).

960, 2020-08-01:
I am not able to get anything but the 1-2-2-1-1-1 ratio with any firmware or any settings.

The only way I get it right is to set these times in hwHandleShaftSignal.

In trigger_decoder.cpp, it gets those times no matter what..

mck1117, 2020-08-01:
[image]

960, 2020-08-01:
Whats been keeping mine from stopping are the change to the gap-check.

it always stopped when there was a delay too close to the gap, and firmware thougt it was the gap.

Now it doesnt "conclude" with seeing a gap until knowing what comes after the gap.

960, 2020-08-01:
I took one of the versions that didnt stop the engine, and the debugger shows the right ratios..
Tried a newer version, same trigger-folder and get the wrong ratios.

It cant figure where/how/why it gets those 1-2-2-1 duratons..

mck1117, 2020-08-02:
have you touched this code at all? `trigger_central.cpp:291`:

```
	// for effective noise filtering, we need both signal edges, 
	// so we pass them to handleShaftSignal() and defer this test
	if (!CONFIG(useNoiselessTriggerDecoder)) {
		if (!isUsefulSignal(signal PASS_CONFIG_PARAMETER_SUFFIX)) {
			/**
			 * no need to process VR falls further
			 */
			return;
		}
	}
```

960, 2020-08-03:
No, I had the whole trigger-folder in sync all the time.

I moved the LogTriggerTooth(signal, timestamp PASS_ENGINE_PARAMETER_SUFFIX);

To after that function now, and this is how it get those times:

[image]

960, 2020-08-03:
// for effective noise filtering, we need both signal edges, 
	// so we pass them to handleShaftSignal() and defer this test
	if (signal == SHAFT_PRIMARY_RISING) {
			return;
		}
#if EFI_TOOTH_LOGGER
		LogTriggerTooth(signal, timestamp PASS_ENGINE_PARAMETER_SUFFIX);
#endif /* EFI_TOOTH_LOGGER */

If I "invert" that function, the times get right:

[image]

960, 2020-08-03:
With VR, there will always be a signal in the gap.

So I guess this is all down to polarity?

I see bosch capture these times at falling edges..

mck1117, 2020-08-03:
invert your trigger input polarity

960, 2020-08-03:
Where are the trigger-simulator signal feed?

960, 2020-08-04:
Inverting it looks better.
So correctly wired sensor = inverted?

Using the rising edge are not exactly correct.
Scoping with other ecu's shows opposite..

960, 2020-08-05:
Any way, this gap-check should be improved to avoid it seeing a gap at the first wider tooth it sees.
It even sees a gap with inverted VR and 2-2-1-1 ratio:

```
for (int i = 0;i<GAP_TRACKING_LENGTH;i++) {
				bool isGapCondition = cisnan(triggerShape->syncronizationRatioFrom[i]) || (toothDurations[i] > toothDurations[i + 1] * triggerShape->syncronizationRatioFrom[i]
					&& toothDurations[i] < toothDurations[i + 1] * triggerShape->syncronizationRatioTo[i]);
				isSync &= isGapCondition;
			}
```

mck1117, 2020-08-05:
On 60-2, by default we look for a gap with ratio from 2.25 to 3.75.  That could probably be reduced.

960, 2020-08-06:
In VAG trigger it says 1.6 to 4.

Setting it to more than 2 and the simulator doesnt start.

Using rising edge for VR is just plain wrong.

The last tooth before gap will always be a falling edge

960, 2020-08-06:
With custom missing teeth trigger it doesnt start without inverting input..

The VAG 60-2 are totally wrong

mck1117, 2020-08-06:
> With custom missing teeth trigger it doesnt start without inverting input..

correct, you need inverted input to be set.

960, 2020-08-06:
Tested with untouched trigger-folder, and with custom toothed wheel it died after around 500 m :-)

Changed to VAG-Decoder, and it stops at higher rpm's again..

Tooth logger stopping with vag-decoder

[2020-08-06_20.47.04.csv.zip](https://github.com/rusefi/rusefi/files/5037493/2020-08-06_20.47.04.csv.zip)

mck1117, 2020-08-06:
> Tested with untouched trigger-folder, and with custom toothed wheel it died after around 500 m :-)

So it worked, is what I'm hearing.  Log of this happening? Logic analyzer?

mck1117, 2020-08-07:
this is probably a duplicate of #988

960, 2020-08-07:
Ah, so it has been discussed before..

I have never seen the wheel with the wide tooth, but it's wrong to call it missing teeth :-)

At least the vag-trigger decoder is wrong, as it works inverted.
In tooth-logger as well it looks for two wider teeth either way.

Cant the trigger-stuff be put into classes as well?
Trigger_central, trigger_structure and trigger_decoder are pretty massive and confusing.

960, 2020-08-09:
With this tiny change, it seems to be running fine:

```
for (int i = 0;i<GAP_TRACKING_LENGTH;i++) {
		bool isGapCondition = cisnan(triggerShape->syncronizationRatioFrom[i]) || (toothDurations[i + 1] > toothDurations[i + 2] * triggerShape->syncronizationRatioFrom[i]
		&& toothDurations[i + 1] < toothDurations[i + 2] * triggerShape->syncronizationRatioTo[i] && toothDurations[i] * triggerShape->syncronizationRatioFrom[i] < toothDurations[i + 1]);
		isSync &= isGapCondition;
	}
```

mck1117, 2020-08-16:
> I have never seen the wheel with the wide tooth, but it's wrong to call it missing teeth :-)

Volvo and BMW with flywheel VR sensors have a long tooth.  58 notches milled in to the flywheel, and a long part without a milled spot.

Either way, the two "flavors" of missing tooth produce identical signals, but inverted from one another if sensed with the same sensor.

960, 2020-08-17:
Ah, ok.. Never seen that one :-)

I have not had time to look any more at this lately.

Do you have any idea what can cause the delayed scheduler thing?

I think the whole trigger part is one huge mess, so could be cool to see how you would done it.. 
I still think you are a hell of a programmer Matt, and some of the issues you have solved before have been heavy :-)

960, 2020-08-20:
Almost lost interest for this stuff/all computer-stuff after holiday, but had a look at some of your branches..

You have actually done a lot cool stuff Matt :-)

Just had a drive with both your gaptracker and software-knock 👍 

Tested with real knock, but the gauge showed values between - 50 to - 30..

What kind of value should the variable knockBandCustom have?

960, 2020-08-20:
And can you please start to add your name as author in files?
In firmware, even if the whole file-content are replaced, they doesn't have your name.

It's useful to see who wrote the code..

mck1117, 2020-08-20:
> And can you please start to add your name as author in files?

No, because git handles that. https://git-scm.com/docs/git-blame

960, 2020-08-20:
Ok, never seen that. I'll test it :-)

But that requires git? and are gone if this repo are deleted?
When working in eclipse or any editor you cant see it..

960, 2020-08-20:
Closed source at 500 sold ecus might sound familiar for some ;-)

960, 2020-08-21:
Your software-knock actually seems to work!!
How to add a second sensor?

mck1117, 2020-08-21:
@960 can you share a datalog, tune, and the ignition table you used to induce knock?

960, 2020-08-21:
I'll try later today.

I have a KnockLite connected, and was just comparing the gauge with that one

mck1117, 2020-08-26:
@960 have a chance to grab a log of knock occuring?

960, 2020-08-27:
I can try to get it done this weekend.

Have been focusing at the spi-fram at the spare time lately, and guess what..

It's working!! :-)

I never tought that I would be able, so this is a huge surprise..

What kind of values should I expect at the gauge?
(I am not sure which of the two pins that go to the connected sensor. One of them are connected to the knocklite)

mck1117, 2020-08-28:
the exact values depend on your engine - what's important is the dynamic range between the conditions of "knock" and "not-knock".

It's a log scale, so -10 is actually 10x the intensity of -20.  I think with the dynamic range of our ADC, it should be able to sense from around -50 to +8.

960, 2020-08-29:
Matt are the only one who knows the firmware and behave adult, so a chat room I guess :-)

With the spi-fram I have a issue I cant figure..

It works correct first time after flashing when values are sent from TS.

But after reset, debugchannels doesnt work, and when I hit burn I get crc-error message.

I have updated the commands and tunerstudio.cpp

Do you see anything not normal here?

[image]

960, 2020-08-30:
Solved :-)

Now I can start testing the knock..

I am not going to generate any heavy knock, but I'll try to make it see what the knocklite does
(It has a led-scale)

960, 2020-08-30:
What's this in the knock-file?

palClearPad(GPIOD, 2);

mck1117, 2020-08-30:
> What's this in the knock-file?
> 
> palClearPad(GPIOD, 2);

please run from master

that was testing logic that's now removed

960, 2020-08-31:
I just had a drive with the spi-fram and knock, but the ecu bricked after loading a tune..

Was sure it was the f-ram, so have been trying to reproduce the brick..

Just found the reason, and it was the software-knock triggering a hardfault..

Turned out this value was 600, even I have a maxvalue of 30 in TS..

Just added this:

if (CONFIG(knockBandCustom) > 30) {
	CONFIG(knockBandCustom) = 30;
	}

960, 2020-08-31:
Now it trips hardfault when it comes here..

addChannel("Baro Press", engineConfiguration->baroSensor.hwChannel, ADC_SLOW);

960, 2020-08-31:
Applied your fix in PR, and now it makes it to: 

 m_thread = chThdCreateStatic(m_threadstack, sizeof(m_threadstack), m_prio, StaticThreadTaskAdapter, this);

960, 2020-09-01:
Want some knock?

Made some "red" knock at the beginning, some normal driving then some "yellow" knock at the end

[2020-09-01_21.44.00.mlg.zip](https://github.com/rusefi/rusefi/files/5158502/2020-09-01_21.44.00.mlg.zip)

960, 2020-09-01:
[image]

Looks like this can actually be used for something with some tuning

mck1117, 2020-09-01:
Ooh that does actually look like knock! "red" and "yellow" mean that you were comparing with your knocklite?

What did you have the knock frequency set to?

960, 2020-09-01:
Yes, its got a led-scale. It's very accurate compared to using headset.

 I think I had 12 in that setting.

The build had some other bugs, so I didnt get to test so much.
Will test some more when I got a better working firmware :-)

mck1117, 2020-09-01:
cool - sounds good.

btw you do have per-cylinder info:
[image]

Might be an interesting test to add some advance/retard to one cylinder, and watch that one knock

[24 more replies]

commit 74f8efd7 (2020-07-30): Trigger gap-check toothDuration array samples time at any event? fix #1658