rusefi/rusefi #1665: migrate VSS to universal event capture

rusefillc, 2020-08-03:
VSS is currently using ICU input capture

we shall migrate VSS to use universal abstract layer

and we shall migrate Frankenso and MRE to PAL :)

rusefillc, 2020-08-16:
See #1668

rusefillc, 2020-08-16:
We need PAL option added in #1689 to be emabled in MRE-F4 config

Once PAL VSS is enabled we need to edit mapping.yaml or else to have more pins available for VSS input

rusefillc, 2020-08-16:
current status:
we now have PAL VSS implementation

we noe need to switch MRE F4 from ICU VSS mode to PAL VSS mode

if needed we can also switch MRE F4 to PAL trigger input from current ICU trigger input

stefanst, 2020-09-03:
Can we have a list of analog inputs that can potentially be used for VSS input without interfering with CAM/CRANK sensing?

rusefillc, 2020-09-03:
@stefanst please see https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/microrusefi/mapping.yaml

cam/crank is GPIOA_5 and GPIOC_6

additional inputs could be anything not ending with 5 or 6, so

safest bet
  # PA7
  EFI_ADC_7: "31 - AN volt 3"

another option but we would have to confirm that AUX reuse low side is not interfering
# PC4 - also AUX REUSE
  EFI_ADC_14: "32 - AN volt 6"

rusefillc, 2020-09-06:
relates to #1533

mck1117, 2021-11-29:
vss uses frequencysensor now

closed 2021-11-29 by mck1117