rusefi/rusefi #1673: NA6 MAF tune

rusefillc, 2020-08-06:
We now have canned NA6 MAP tune @stefanst please validate build-in MAP tune does it match your expectations?

We need NA6 MAF tune in addition to MAP tune

rusefillc, 2022-08-22:
@BeerMoneyMotorsports I am looking at FB and it makes me wonder "would it help users if we make progress with some or all of hellen hard-coded tunes?"

it _feels_ like there is a huuuuge oportunity somewhere in that department. I could be very wrong.

I also remember someone smart suggesting that I shall simply shut up :)

Any change that @stefanst could glance over all four canned tunes? Any chance that @stefanst could glance over "as shipped with BMM" tunes which I wonder where are those even coming from? Does each NB1 unit get shipped with same tune same FW? current snapshot/adjusted tune? else?

BeerMoneyMotorsports, 2022-08-23:
I agree, it would be a big step forward.  I used the uploaded 96/97 tune as a base for someone but i had better idle settings available.  With some minor tuning they report "it runs as good as stock and is far better than the MS2 it replaced".  I will get that tune uploaded to REO and we can start there.

BeerMoneyMotorsports, 2022-08-23:
96/97 tune
https://rusefi.com/online/view.php?msq=1179

BeerMoneyMotorsports, 2022-08-23:
Edit- it has nb1 injectors so the previous injector settings will need to stay

rusefillc, 2022-08-23:
@BeerMoneyMotorsports open question what's the best way forward. Do I need to invest time to have better tools or should we see if progress could be made manually

We need a collection of FOUR fresh default tunes. It kind of pains me to image you messing with four ECUs to grab default tunes from each version :( Sounds like automation is needed damn

elmepro, 2022-08-23:
fuel table must be tuned on dyno, its impassible to tune correctly on road, im not saying its impassible but if you use brakes on road then maybe you will get good result, after that you can change your brakes ;d

BeerMoneyMotorsports, 2022-08-23:
@elmepro esto es para melodías de referencia y se entiende como un punto de partida.  Debería poder arrancar y dejar el coche en ralentí. Posiblemente ser capaz de conducir el coche.
this is for baseline tunes and meant as a starting point. It should be able to start and idle the car.  Possibly be able to drive the car.

BeerMoneyMotorsports, 2022-08-23:
@rusefillc i will link a tune and a difference report(snapshots ok?) with the current base tune. 

Large question is appropriate injector data, specifically dead times, for common injectors.  This would be for the base tune(stock injectors) and for aftermarket.

rusefillc, 2022-08-23:
We have multiple milestones of qulity. I've adjusted titme and summary hopefully this would get us all on the same page.