rusefi/rusefi #1674: M111 PnP dream, connector breakout board

rusefillc, 2020-08-06:
We need OEM connector footprint, there are two ways to figure it out

Plan A, easier: google/ask on relevant forums/else, find engineering drawing of connector based on part number

Plan B, harder: dis-solder and measure. What is the cheapest ECU on eBay which uses the right 145 pin connector?
Can we acquire compatible ECU, solder out the connector and ship it to @jharvey for measurement?

duhafnusa4, 2020-08-06:
[image]
[image]

rusefillc, 2020-08-06:
We need more details on - wire side info does not help :(

[image]

duhafnusa4, 2020-08-06:
[image]
[image]

rusefillc, 2020-08-07:
A 023 545 97 32 for $17 might use same connector?

rusefillc, 2020-12-26:
https://www.ebay.com/itm/128-pin-ECU-connector-Mercedes-breakout-PCB-M111/333791544427

closed 2026-01-28 by rusefillc