rusefi/rusefi #1676: 80 pin VAG connector breakout board

rusefillc, 2020-08-06:
We do not know anyone with this connector but we have forum inquiry https://rusefi.com/forum/viewtopic.php?f=4&t=1818
 
I have these connectors new for sale so selling with connector would be a nice publicity item

jwrb, 2020-08-06:
Models and pin latouy for this connection can be found on the TE Connectivity website:

https://www.te.com/usa-en/product-8-1393476-0.html

jwrb, 2020-08-06:
I already brought the connector into KiCad to start making a PCB, but it's been a few years since I touched any PCB Cad software so I'm a little lost.

The files I used are attached.
[8-1393476-0_KiCad Libraries.zip](https://github.com/rusefi/rusefi/files/5037703/8-1393476-0_KiCad.Libraries.zip)

[image]

rusefillc, 2020-08-13:
@jwrb I've pushed both files into rusEFI library repository

Any progress on the adapter board?

jwrb, 2020-08-25:
@rusefillc It's been years since I dabbled with PCB design. Still working my way through it. Would be very nice to have a layout.

I'd like to get it designed so I can have some PCB's made for my own builds.

rusefillc, 2020-10-05:
Aborting

closed 2020-10-05 by rusefillc

jwrb, 2020-10-07:
@rusefillc 
I've gotten in contact with a couple of PCB designers regarding this. I'll be commisioning a PCB layout for this connector.

Once complete, I will add it to the repo, and submit a PR.

Thanks

rusefillc, 2020-10-07:
I've actually already paid someone to make a breakout board - just pushed, see associated commit. 

Between you contributing the footprint I have to make the board public but I will probably not publish official gerbers for this board.

jwrb, 2020-10-07:
> I've actually already paid someone to make a breakout board - just pushed, see associated commit.
> 
> Between you contributing the footprint I have to make the board public but I will probably not publish official gerbers for this board.

Oh great! That's fab news! I'll have to see about sorting a few boards for my personal use!
Thank you! :)