rusefi/rusefi #1701: F7 firmware issues with TS connection while transferring big data blocks

shadowm60, 2020-08-18:
There are issues with packets >256 bytes.

In code the following condition is failing:
incomingPacketSize > (sizeof(tsChannel->crcReadBuffer) - CRC_WRAPPING_SIZE

The workaround to have a usable F7 firmware is to change in TS ini file:
blockingFactor = 128 ; max chunk size

Tested with    blockingFactor = 249 ; max chunk size
This means data + 7 frame bytes -> total 256 bytes.

Works fine.

mck1117, 2020-08-18:
[image]

960, 2020-08-20:
With these changes my TS-connection fails (F7) :-)

I probably have different settings somewhere.. No worries :-)

mck1117, 2020-08-20:
Remember, we can't help you without you using official firmware and ini.  @960, please keep it on topic in this issue.

960, 2020-08-20:
Ah, remember me complaining about beeing unable to test with official firmware and .ini because of TS comms :-)

Just remembered it was impossible to upload settings..