rusefi/rusefi #1707: Critical error: Pin "PA0"

Cartoonian, 2020-08-20:
I was following along with the software guide provided [here](https://rusefi.com/wiki/index.php?title=Manual:Software:User/en) and I started up the console to this message:

EngineState: CRITICAL error: Pin "PA0" required by "TPS 1 Secondary" but is used by "MAF" 20200820@frankensoNA6

The logs contain this message repeating:

2020-08-20_17_07_39_910<EOT>: Error processing version [20200820@25130source MIATA_NA6_VAF 412]

I have the Frankenso with the jumpers done for the na95.
Sorry if it's something simple.

rusefillc, 2020-08-20:
you have two sensors - "TPS 1 Secondary" and "MAF" - trying to read from same PA0 that's expected to fail

i believe there is something about upgrading firmware and #1708 is a usability issue here

Cartoonian, 2020-08-21:
> 
> 
> you have two sensors - "TPS 1 Secondary" and "MAF" - trying to read from same PA0 that's expected to fail
> 
> i believe there is something about upgrading firmware and #1708 is a usability issue here

Thank you for the fast reply!

Should I have not updated the firmware? 
What would I need to do to fix this? 
Could I alter the code myself?

rusefillc, 2020-08-21:
what software to you use to change settings? Tuner Studio? Please find "TPS 1 Secondary" and set it to "NONE"

no code changes are needed
we recommend using latest software

Cartoonian, 2020-08-21:
I'm using tuner studio, but the option it gives is "Disabled." When I select that, burn it, and reset the same error appears.

rusefillc, 2020-08-21:
can we move this to rusefi forum?

rusefillc, 2020-08-21:
can you reset to frankenso na6 settings? there is a button in popular vehicles

Cartoonian, 2020-08-21:
Starting a thread with the same name in software.

mck1117, 2020-08-21:
I think we can close this as a duplicate of #1708