rusefi/rusefi #1731: Wake-on-can

rusefillc, 2020-08-28:
https://github.com/rusefi/rusefi/wiki/VolkswagenPassatB6 is the first rusEFI vehicle with wake-on-CAN ECU power management.

rusefillc, 2020-08-28:
@mck1117 says "typical power budget for always-on things like the ECU is something like 0.00005A (50uA)"

mck1117, 2020-08-28:
Wake-on-CAN requires a can tranceiver with "CAN partial networking (PN)".

https://www.microchip.com/design-centers/can/can-transceivers/can-bus-partial-networking-transceivers

> Microchip CAN PN transceivers offer very low power consumption in Standby and Sleep modes and support ISO 11898-2:2016 compliant CAN Partial Networking by means of a selective wake-up function. A CAN frame decoder evaluates the bus traffic and checks for a matching frame that has been configured into registers via the SPI. 

tl;dr they include a cut-down CAN controller that can listen to a particular CAN ID while in a low power mode with the main CPU turned off.

rusefillc, 2020-08-28:
Does TLE8888 offer this @mck1117 @dron0gus ?

mck1117, 2020-08-28:
iirc it can wake up on the presence of CAN traffic, but not check for a particular ID

mck1117, 2020-08-28:
we would have to make some architectural changes to MRE to get remote wakeup to work propertly (and actually get in to a low power state when not woken up)

shadowm60, 2020-08-28:
Usually power supply architecture is different.
For Can we would need something like: TJA1043 – High speed CAN transceiver with Sleep Mode and diagnostics
This is powered by KL30 and it will rise Inhibit pin to 12v when it detects can activity. This inhibit pin would need to start the power supply so micro could wake up.

Note that power supply would need OR for wakeup/keep alive so that way as soon as the micro wakes up and validates boot (ex. checks that inhibit is high) it would keep the power supply alive.

When going to sleep, the can would shut down (usually Network management messages are the last, then there is silence on CAN), so if this is the case the micro would need to place TJA in sleep mode and after that release the power supply. (inh is low, micro power supply is low, so the micro would run out of power). -> ECU in sleep mode.

The wake up part is not that tricky, the going to sleep is a bit tricky.
Usually NM would announce that network is ready to sleep, some ecus might request to keep transmitting last messages, till they are ready to go to sleep, after that Sleep indication and everybody would go to sleep. In this critical time window we should not transmit messages so the whole network would go to sleep.

Some standards, but this is applicable for new cars and they are sometimes a bit modified for OEM specific stuff
https://www.autosar.org/fileadmin/user_upload/standards/classic/3-0/AUTOSAR_SWS_CAN_NM.pdf

rusefillc, 2021-12-25:
there are about 70 MIKROE-2900 dev boards with ATA6570 chip for sale https://octopart.com/search?q=MIKROE-2900&currency=USD&specs=0

rusefillc, 2021-12-25:
Wow there is tiny stock of TJA1145 https://octopart.com/search?q=TJA1145&currency=USD&specs=0

rusefillc, 2021-12-25:
TLE9255W none :(
https://octopart.com/search?q=TLE9255W&currency=USD&specs=0

a few of MIKROE-4062 are available

rusefillc, 2024-01-14:
> Wake-on-CAN requires a can tranceiver with "CAN partial networking (PN)".

Not really, you need fancy transceiver only if you want to wake up on _specific_ CAN message. See https://github.com/andreika-git/hellen-one/tree/master/modules/mega-mcu144 it seems to wake up on _any_ CAN with a regular transceiver.

Overall rusEFI has at least basic wake-on-CAN now.

closed 2024-01-14 by rusefillc