rusefi/rusefi #1733: OBD CAN sensors

rusefillc, 2020-08-29:
For TCU, BCU, piggyback and even dual rusECU use-case we shall read sensor data from CAN

At the moment we have something for this:
CanSensorBase - hopefully could be used for ODB-II sensor input
nasty canMap for custom rusEFI CAN

We need support for many/most basic sensors from RPM to timing angle

mck1117, 2020-09-01:
should we really exert effort on this when we could pretty easily support (with almost new code written) reading broadcast CAN sensors? that would cover most piggyback scenarios, and would certainly cover multi-rusefi scenarios.  The performance is also much better - you don't have to wait for a response

rusefillc, 2020-09-01:
You bring valid points for sure but I believe that OBD CAN sensors are a very cheap way to use rusEFI as flexible BCM for non-rusEFI installations, I would like to POC that

rusefillc, 2020-09-08:
bare minimum implementation works

closed 2020-09-08 by rusefillc