rusefi/rusefi #1739: Fuel pump PWM

rusefillc, 2020-08-31:
Do we need fuel pump PWM? Would GP PWM be enough?

nilayksrivastava, 2020-08-31:
It would be great to have Fuel pump PWM control for returnless fuel sending units.

rusefillc, 2020-08-31:
@nilayksrivastava is this something you can contribute to by any chance?

nilayksrivastava, 2020-08-31:
I would love to. I am considering having that functionality for my motorcycle using MRE. I would need to benchmark some possible use case scenarios for existing hardware in other vehicles.

nilayksrivastava, 2020-09-02:
[image]

Here's what I have been thinking for the basic PWM logic. I have a standard feed forward system since I haven’t yet come across a system with a fuel pressure sensor on the low pressure fuel side (Benchmarked a VW BPY 2.0 FSI Fuel sending unit).
The blocks highlighted in yellow are values that can be calibrated. I may need to add logic for taking into consideration system response characteristics. Values highlighted in green/red are inputs, the orange highlighted are outputs.
1. Base Duty cycle Map holds a 2-D lookup table for base pump duty cycle % as per speed vs load (acc pedal position) values. I'm also considering speed vs injected fuel quantity as an alternative.
2. Fuel Temp correction is a simple 1-D array of gains for correcting duty cycle and low side fuel press under conditions such as hot soak. This map could be kept optional to use in case of absence of fuel temp sensor in certain systems, but I personally feel it would be a valuable option to have especially for future GDI capability addition.
3. Pre start duty cycle commands a base duty cycle % value for fuel system priming during key on and post engine stall with a calibratable value for pump off timeout.
4. PumpPrime Time out_ms is a calibratable value for controlling the time for which the pump remains on after an engine stall event or an ignition on event. 5000 ms is set as default.
5. ENG_RUN_CRANK_BOOLEAN is a value that alters between 0 and 1. (0-False, 1-True). This value should set to 1 when the engine is cranking to maintain optimum fuel pressure.
6. IGN_ON_BOOLEAN is also a value that alters between 0 and 1. (0-False, 1-True).

I will try converting this to C code during the coming weekend. Due to my ME background I am not very comfortable with C/C++ but I'm trying to learn. I have a PWM based standard fuel pump control module which could be put to use for testing.

rusefillc, 2020-09-02:
@OrchardPerformance @mck1117 ^^^

mck1117, 2020-09-02:
Do we really need that involved a controller?

AFAIK, the following combinations of fuel system exist:
1. Return system, manifold referenced
2. Return or returnless system, atmospheric referenced (NB Miata, late 00s GM, etc)
3. Returnless system, unregulated, with pressure sensor (Recent GM, etc)
4. GDI, which is usually option 3 that just feeds an independently controlled mechanical pump

Options 1 and 2 don't need a complicated controller: the duty just needs to be high enough that you're pumping more than the engine needs so the regulator can do its job.

Option 3 needs a controller, but you can be pretty lazy about it, since you can correct the injector flow with the sensed pressure on the fly.  If the fuel pressure is lazy to catch up, that's no problem, since you can correct with injection duration in the mean time.

mck1117, 2020-09-02:
Option 1/2 can be implemented today using GP PWM: Switch the pump controller's power with a relay (as if it was an on-off pump), but add a PWM signal from gppwm to control duty cycle.  At zero rpm, run whatever duty you want for priming.  When running, run whatever duty is necessary to get enough fuel at that particular rpm/load operating point.

nilayksrivastava, 2020-09-02:
> Do we really need that involved a controller?
>
>
>
> AFAIK, the following combinations of fuel system exist:
>
> 1. Return system, manifold referenced
>
> 2. Return or returnless system, atmospheric [...]

I was wanting to implement this exact solution on a return less system for my application. Regarding fuel injected quantity correction due to suboptimal fuel pressure by increasing injector pulse width there could be problems with spray pattern distortion which could be undesirable in especially GDI applications.

mck1117, 2020-09-02:
> there could be problems with spray pattern distortion which could be undesirable in especially GDI applications.

Sure - but the priority is accurate fueling above all else.  If you have the choice to correct fueling with pressure or not, of course you do it, because otherwise you'll end up chasing transients that the fuel system can't physically keep up with.

Having datalogged fuel pressure while driving an OEM GDI setup, you'd be surprised how lazy the fuel pressure control is.  Also remember that all of the flow dynamics through the injector are proportional to the square root of the pressure, so +-10% pressure is really only +-4% flow.

nilayksrivastava, 2020-09-02:
> At zero rpm, run whatever duty you want for priming. When running, run whatever duty is necessary to get enough fuel at that particular rpm/load operating point.

I didn't read this reply earlier. That's exactly more or less what the above logic is doing with the exception of an added fuel temp based gain for the duty cycle.

nilayksrivastava, 2020-09-02:
> Sure - but the priority is accurate fueling above all else. If you have the choice to correct fueling with pressure or not, of course you do it, because otherwise you'll end up chasing transients that the fuel system can't physically keep up with.

That's exactly my intent for having this kind of a control. Environmental variations can induce all kinds of variations in fuel/air handling systems. Therefore for my setup (PFI) I would want to do a separate benchtop injector calibration for various speed load points at different fuel pressures (3-6 bar). Maintaining the cone angle is also of critical importance to prevent unnecessary manifold wetting. Fuel injector dynamics will also influence the injection quantity on a shot by shot mm3/str basis which I'm interested in documenting. I have 2 Bosch EV14 brand new injectors to start with.

> Having datalogged fuel pressure while driving an OEM GDI setup, you'd be surprised how lazy the fuel pressure control is. Also remember that all of the flow dynamics through the injector are proportional to the square root of the pressure, so +-10% pressure is really only +-4% flow.

I have never really worked with/on GDI but I do deal with heavy duty diesels on a daily basis at work, I have to say that the fuel system calibration on these is pretty sharp, though again the slightest of the environmental variations can throw things off balance.

rusefillc, 2021-11-18:
Lua?

closed 2021-11-18 by rusefillc