rusefi/rusefi #1756: Proteus Tunerstudio does not allow selecting digital inputs for AC switch pin setting

ZHoob2004, 2020-09-05:
Unable to select digital inputs 1-6 as options for ac switch pin. As-is, there is no way to configure a low-side a/c switch.

rusefillc, 2020-09-06:
Works for me

[image]

rusefillc, 2020-09-06:
https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/proteus/mapping.yaml looks good

rusefillc, 2020-09-06:
this is more than just a TS problem

at the moment we have a mess with input switches - some are implemented as ADC inputs not digital inputs for historical reasons

rusefillc, 2020-09-07:
```
	adc_channel_e hipOutputChannel;
	adc_channel_e acSwitchAdc;A/C button input handled as analogue input
	adc_channel_e vRefAdcChannel;
```

acSwitchAdc would have to be changed into acSwitch without ADC and actual pin polling code would need to be changed

chuckwagoncomputing, 2020-09-09:
Is anyone using acSwitchAdc? Can I get rid of it without messing anyone's config up?

ZHoob2004, 2020-09-09:
The fsio code I got from @stefanst uses acSwitchAdc

rusefillc, 2020-09-09:
proper fix would be to convert all adc switches into digital inputs

na6 pnp uses acSwitch

chuckwagoncomputing, 2020-09-09:
Why was it implemented as ADC in the first place?

rusefillc, 2020-09-09:
at the time that was the quickest way to implement an input

chuckwagoncomputing, 2020-09-11:
This can be closed

rusefillc, 2020-09-11:
This is now fixed, ac input is now digital not adc

closed 2020-09-11 by rusefillc