rusefi/rusefi #1767: Plugin browsing of SD card logs limited to 16 files

stefanst, 2020-09-07:
It appears that the plugin only lists 16 files. And it also seems to miss the 2nd digit. So in my list the first 9 logs have sensible names, but from log 10 on they are all named "rusefi_1.mlg". assuming that mlg files with numbers of 20 and higher would be listed as "rusefi_2.mlg"

[image]

stefanst, 2020-09-07:
This naming infidelity also causes an inability to download or delete files with numbers 10 or higher.

rusefillc, 2020-09-07:
MS3 protocol which we do not HAVE to use but WANT to use uses legacy DOS short 8+3 file names

in light of this current RE log files prefix was changed from "rusefi_" to just "re_"

16 files is my laziness, this could be expanded but for now one can just remove files until one has not more than 16?

stefanst, 2020-09-07:
Wow, I;m not sure I have previously run into the DOS 8+3 restriction in this millennium. Last millennium? Yes.

And sure 16 files is acceptable for now and laziness is, at times, forgivable.

mck1117, 2021-03-15:
significantly mitigated by completing #140 via #2349

mck1117, 2021-09-28:
Fixed by removing weird TS SD protocol, replaced with USB MSD.

closed 2021-09-28 by mck1117