rusefi/rusefi #1768: Document the benefits of ConfigOverrides or maybe get rid of it

rusefillc, 2020-09-07:
I am not getting the benefits of ConfigOverrides but at the moment it seem to be affecting me negatively

Please document the value proposition or maybe let's remove this?

mck1117, 2020-09-28:
what does "affecting you negatively" mean?

rusefillc, 2020-09-28:
"affecting me negatively" as in "breaks my brain and makes my eyes bleed"

mck1117, 2020-10-01:
It is something that's important - in the vein of #1834, we either need to:
1. provide a config option in TS for a feature
2. force the config to the desired state for that feature

1 isn't ideal, as it clutters up TS with options that are at best confusing, and at worst will cause undesired behavior (why should a user be able to select battery voltage channel on MRE/Proteus/etc? We know what pin it's on!)

That leaves us with option 2: overriding config at compile time with hard-coded config on a per-board basis.

#1840 is some sort of progress.

PR #2508 (2021-03-31): make board config override actually usable
Simply calls a function after config load to let the board force particular config options.  These options are ones that are intrinsic to the board, and shouldn't be changable: ETB pinout, SD card pinout, LEDs, analog temp bias resistors, etc.

Fixes #1768
closed 2021-03-31 by rusefillc