rusefi/rusefi #1773: GHA unit_tests does not fail while it should

rusefillc, 2020-09-08:
https://github.com/rusefi/rusefi/commit/cab4cd0df2c82728d20d8659f32812d178bb4e26 is probably bad? yet jenkins fails while GHA does not?

```
Compiling logicdata.cpp
Compiling main.cpp
Compiling tooth_logger.cpp
Compiling log_field.cpp
../firmware/console/binary/tooth_logger.cpp: In function 'int copyCompositeEvents(CompositeEvent*)':
../firmware/console/binary/tooth_logger.cpp:62:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for (int i = 0;i < NextIdx;i++) {
                 ~~^~~~~~~~~
Linking build/rusefi_test
build/obj/debounce.o: In function `ButtonDebounce::updateConfiguration()':
/cygdrive/c/Program Files (x86)/Jenkins/workspace/unit_tests/unit_tests/../firmware/hw_layer/debounce.cpp:58: undefined reference to `brain_pin_markUnused'
/cygdrive/c/Program Files (x86)/Jenkins/workspace/unit_tests/unit_tests/../firmware/hw_layer/debounce.cpp:59: undefined reference to `getInputMode'
/cygdrive/c/Program Files (x86)/Jenkins/workspace/unit_tests/unit_tests/../firmware/hw_layer/debounce.cpp:59: undefined reference to `efiSetPadMode'
collect2: error: ld returned 1 exit status
make: *** [rules.mk:155: build/rusefi_test] Error 1
/cygdrive/c/Program Files (x86)/Jenkins/workspace/unit_tests/unit_tests
```

rusefillc, 2020-09-08:
My local build confirm the error - on my Windows under Cygwin I get

```
../firmware/console/binary/tooth_logger.cpp: In function 'int copyCompositeEvents(CompositeEvent*)':
../firmware/console/binary/tooth_logger.cpp:62:19: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'long long unsigned int'} [-Wsign-compare]
  for (int i = 0;i < NextIdx;i++) {
                 ~~^~~~~~~~~
Linking build/rusefi_test
build/obj/debounce.o: In function `ButtonDebounce::updateConfiguration()':
C:\stuff\rusefi\unit_tests/../firmware/hw_layer/debounce.cpp:58: undefined reference to `brain_pin_markUnused'
C:\stuff\rusefi\unit_tests/../firmware/hw_layer/debounce.cpp:59: undefined reference to `getInputMode'
C:\stuff\rusefi\unit_tests/../firmware/hw_layer/debounce.cpp:59: undefined reference to `efiSetPadMode'
collect2.exe: error: ld returned 1 exit status
make: *** [rules.mk:155: build/rusefi_test] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

12:03:13 Build Failed. 4 errors, 103 warnings. (took 5m:4s.73ms)

```

```
C:\stuff\rusefi>gcc -v
Using built-in specs.
COLLECT_GCC=C:\Program Files\mingw-w64_x86_64-8.1.0\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/Program\ Files/mingw-w64_x86_64-8.1.0/mingw64/bin/../libexec/gcc/x86_64-w64-mingw32/8.1.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../../../src/gcc-8.1.0/configure --host=x86_64-w64-mingw32 --build=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --prefix=/mingw64 --with-sysroot=/c/mingw810/x86_64-810-posix-seh-rt_v6-rev0/mingw
64 --enable-shared --enable-static --disable-multilib --enable-languages=c,c++,fortran,lto --enable-libstdcxx-time=yes --enable-threads=posix --enable-libgomp --enable-libatomic --enable-lto --enable-graphite --en
able-checking=release --enable-fully-dynamic-string --enable-version-specific-runtime-libs --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-bootstrap --disable-rpath --disable-win32-registry --disable-nl
s --disable-werror --disable-symvers --with-gnu-as --with-gnu-ld --with-arch=nocona --with-tune=core2 --with-libiconv --with-system-zlib --with-gmp=/c/mingw810/prerequisites/x86_64-w64-mingw32-static --with-mpfr=/
c/mingw810/prerequisites/x86_64-w64-mingw32-static --with-mpc=/c/mingw810/prerequisites/x86_64-w64-mingw32-static --with-isl=/c/mingw810/prerequisites/x86_64-w64-mingw32-static --with-pkgversion='x86_64-posix-seh-
rev0, Built by MinGW-W64 project' --with-bugurl=https://sourceforge.net/projects/mingw-w64 CFLAGS='-O2 -pipe -fno-ident -I/c/mingw810/x86_64-810-posix-seh-rt_v6-rev0/mingw64/opt/include -I/c/mingw810/prerequisites
/x86_64-zlib-static/include -I/c/mingw810/prerequisites/x86_64-w64-mingw32-static/include' CXXFLAGS='-O2 -pipe -fno-ident -I/c/mingw810/x86_64-810-posix-seh-rt_v6-rev0/mingw64/opt/include -I/c/mingw810/prerequisit
es/x86_64-zlib-static/include -I/c/mingw810/prerequisites/x86_64-w64-mingw32-static/include' CPPFLAGS=' -I/c/mingw810/x86_64-810-posix-seh-rt_v6-rev0/mingw64/opt/include -I/c/mingw810/prerequisites/x86_64-zlib-sta
tic/include -I/c/mingw810/prerequisites/x86_64-w64-mingw32-static/include' LDFLAGS='-pipe -fno-ident -L/c/mingw810/x86_64-810-posix-seh-rt_v6-rev0/mingw64/opt/lib -L/c/mingw810/prerequisites/x86_64-zlib-static/lib
 -L/c/mingw810/prerequisites/x86_64-w64-mingw32-static/lib '
Thread model: posix
gcc version 8.1.0 (x86_64-posix-seh-rev0, Built by MinGW-W64 project)
```

rusefillc, 2020-09-08:
GHA now prints it's version

GHA 7 while Jenkins 8

```
Run gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 
```

chuckwagoncomputing, 2020-09-08:
GCC 10.2 on my laptop (Arch) behaves the same as GHA

rusefillc, 2020-09-08:
https://github.com/actions/virtual-environments/issues/103 not sure if relevant

@mck1117 has similar issues before and it seemed like jenkins was right - things were in fact not visible

rusefillc, 2020-09-08:
See https://github.com/rusefi/rusefi/runs/1087166782

is it saying that it has 8.4 but uses 7?

rusefillc, 2020-09-08:
is https://stackoverflow.com/questions/7832892/how-to-change-the-default-gcc-compiler-in-ubuntu  needed to make 8 default?

chuckwagoncomputing, 2020-09-08:
probably. Or maybe just env variables.

rusefillc, 2020-09-08:
that's above my linux grade

rusefillc, 2020-09-08:
@mck1117 says "I can get it to fail on new gcc on linux if I turn off link gc

link gc = delete unused sections (functions)"

closed 2020-09-08 by rusefillc