rusefi/rusefi #1801: ButtonDebounce refactoring

rusefillc, 2020-09-17:
1) please replace DEBOUNCE_INC with #pragma once

2) please add a ~3 line code comment on top of debounce.h explaining the overall problem of electrical debounce to people not familiar with that EE concept. 

3) overall we have a huge confusing with time variables type - we have NT, MS and NS. because of this overall confusion we  are trying different options to reduce confusion - either code comment explicitly stating variable units or variable name suffix. Please make time unit in threshold obvious. 'efitimems_t' is related to this issue exactly but it looks like MS is the only named 32 bit numeric type - the other types in rusefi_types.h are either 64 bit or float.

4) "init(int t, brain_pin_e *p, pin_input_mode_e *m);" please give all parameters proper full names.

5) please review the whole class and add code comments

"bool readValue;" is it "currentValue"? is it logical or electrical value? is this a flag asking to read value or a state or else?

6) it looks like list management could be simplified. at the moment we seem to be adding new elements into the end on the list while it's way simpler to add elements into the head of the list - I assume we do not care for the order of elements in the list but we definitely care for overall code complexity with preference on simpler more readable code.

7) please document why do we go with "initialized" flag and not with constructor, the constructor approach would take one less field?

rusefillc, 2020-09-17:
8. ButtonShiftController needs a better name
TcuGearButtonShiftController maybe? as is, "ButtonShiftController" is too vague. One has to open the file to see references to "gear" but we should be more obvious on what this code is about.

chuckwagoncomputing, 2020-09-17:
9. DebounceButton needs separate start and stop methods for updating configuration, rather than one "update" method.

chuckwagoncomputing, 2020-09-17:
8. is out of the scope of ButtonDebounce refactoring and should be a separate issue.

chuckwagoncomputing, 2020-09-17:
I need 3 explained to me further. What exactly needs to change in ButtonDebounce? Variable names? Comments?

rusefillc, 2020-09-17:
https://github.com/rusefi/rusefi/commit/c37e1c8b70aa2470f39122c7523297eca2acc3ad is an example of simpler (?) linked list implementation

rusefillc, 2020-09-17:
my bad it's actually https://github.com/rusefi/rusefi/commit/6c88cbe5332e4ae2d39f388203201025b19fb11a

As for #3 - let's focus on #1802 first. I believe that #1802 is an amazing example of the actual defect due to lack of clarity, while fixing #1802 we can explore different strategies of how clarity could be added into this area

rusefillc, 2020-09-20:
new issue #1811 in light of broken master

chuckwagoncomputing, 2020-09-29:
I think this can be closed

closed 2020-09-29 by rusefillc