rusefi/rusefi #1802: time units mismatch around start/stop

rusefillc, 2020-09-17:
we have a mess related to time units and @chuckwagoncomputing has totally fallen into the trap with code reviewers not noticing

"int16_t startCrankingDuration;"
"int t"

which one is in seconds? which one is in ms? impossible to tell without reading the code deeper :(

rusefillc, 2020-09-17:
See #1801

we might need to review the whole "efitimems_t" and have explicit type name for more combinations of logic type and storage type

chuckwagoncomputing, 2020-09-18:
I don't know enough yet to even know where to start with this. What types would be replaced by new types?

chuckwagoncomputing, 2020-09-18:
Does efitimems_t already have its own TS field?

rusefillc, 2020-09-18:
I am pretty distracted for 8 more days.

1) please look at current 5-8 type names and confirm that making convention makes sense. There is a chance that a better more more uniform is possible

chuckwagoncomputing, 2020-09-18:
Are you asking for efitimeXX_t types in ConfigDefinition so they can be used in rusefi_config.txt?

rusefillc, 2020-09-18:
yes please try efitimeXX16_t in ConfigDefinition

chuckwagoncomputing, 2020-09-29:
I think this should be closed, and perhaps another opened to address time type issues.

chuckwagoncomputing, 2021-01-02:
@rusefillc What needs to be done on this issue? The mismatch has long since been fixed; the only remaining issue is you wanting to revamp the time type system. I think that if that is something we still want to do, an issue should be opened with a clear statement of what needs to be changed, and this one should be closed.

closed 2021-01-02 by rusefillc