rusefi/rusefi #1803: Output pin init/deinit needs a linked list

rusefillc, 2020-09-17:
at the moment we have too much inconsistent copy-paste around output pin management, need improvements

rusefillc, 2020-09-17:
Same idea as #1800

chuckwagoncomputing, 2020-09-18:
Are you going to finish this, or should I combine this and what I have done in #1800?

chuckwagoncomputing, 2020-09-18:
And what's the reasoning behind a separate RegisteredOutputPin class rather than just including it in OutputPin?

rusefillc, 2020-09-18:
@chuckwagoncomputing yes planning to make progress on this

Separate class for two reasons: I want to start with a subset since not 100% sure how it would apply universally. 2nd reason maybe it would be nice to separate different layers and do OOP. Better OOP would be composition not inheritance maybe in the future, do not know yet.i am just slowly moving forward

chuckwagoncomputing, 2020-09-18:
I'm a bit surprised, as slowly moving forward is not your usual approach. I was hoping to have this functionality in place so I don't have to manage pins manually in #1752, which is otherwise ready.

Could we have a class for pin registration that could be reused by both OutputPin and ButtonDebounce?

rusefillc, 2020-09-18:
I am pretty distracted for next 8 days :( things would probably get faster in 8+ days from now.

rusefillc, 2020-10-05:
Status update: reality is in the way :(

chuckwagoncomputing, 2021-03-11:
Status update? @rusefillc

rusefillc, 2021-03-11:
I think I am finished with what I wanted to implement

closed 2021-03-11 by rusefillc