rusefi/rusefi #1804: F7 version of the MRE has wasted spark selection options issue

OrchardPerformance, 2020-09-17:
MRE F7 greys out the wrong boxes when set to wasted spark. 
[image]

mck1117, 2020-09-18:
not just f7, this part is the same for all configs

mck1117, 2020-09-18:
in your opinion which should it grey out? what's your firing order?

OrchardPerformance, 2020-09-18:
Imo in this case, if we are telling people to use 123 for a 6 cylinder then we should grey out 456 as that would automatically pick the correct outputs.

OrchardPerformance, 2020-09-18:
Firing order is more an issue for the connection of plug leads than the ecu channels chosen. 
Coil one should always be cyl 1 plus what ever cylinder is up at the same time, repeat for 2 and 3.

mck1117, 2020-09-18:
We could also just only allow two wire mode (remove the setting for it), and add a message of "select any ignition outputs you want to use, outputs 360 degrees apart in the firing order will fire together".  There's not really a negative to the firmware flipping pins that are assigned to `NONE`, it's almost free to do.

OrchardPerformance, 2020-09-20:
Just as an addition to this. 
Turns out the ecu was using the highlighted channels, which means the only issue is really that we are different in how we do it for batch fuel and wasted spark. 
Batch fuel highlights outputs 1,2,3 and wasted spark has 1,3,5 causing the confusion

closed 2026-01-28 by rusefillc