rusefi/rusefi #1810: Presets are not user friendly

rusefillc, 2020-09-20:
Users are missing the warning on the presets dialog and still hit 'burn' and get write error. We need a better approach to preset buttons.

rusefillc, 2020-09-20:
See https://rusefi.com/forum/viewtopic.php?f=3&t=1836&p=37875#p37875

rusefillc, 2020-09-20:
We have two options:

1) move presets buttons from TS to plugin or console - this would hide 'Burn' button, the down side is more complexity to even locate the buttons

2) keep the UI as is but make ECU reboot on preset command

mck1117, 2020-11-03:
Here's an idea: when a preset button is pressed, trigger a fatal error on the ECU that says "please reboot ECU to apply preset".  It'll force the user to disconnect/reconnect gracefully instead of the ECU rebooting itself.

rusefillc, 2020-11-03:
I like this idea!!!

mck1117, 2020-11-03:
[image]

Oh yeah, I like this.

PR #1922 (2020-11-04): fatal error + inhibit read on TS preset
this will probably fix #1810

If we pop a fatal error with a message, we can force the user to reboot the ECU.  Likewise, we can inhibit certain functions so that TS can't accidentally overwrite: read, write, and burn will no longer work until the ECU is rebooted.
closed 2020-11-04 by rusefillc