rusefi/rusefi #1811: unit test for start/stop hardware and pin repository

rusefillc, 2020-09-20:
right now master fails HW CI around stop/start pins and repository

our "unit" tests (which are really integration tests) have enough to reproduce this trivial issue at least for debounce hopefully

rusefillc, 2020-09-21:
@chuckwagoncomputing see commits above - can you please rebase on master with these changes? is your PR changing the behavior of this currently failing test?

chuckwagoncomputing, 2020-09-21:
If the cause of the issue is what I think it is, I'm not sure why that didn't trigger it.

chuckwagoncomputing, 2020-09-21:
Oh I know why it didn't. The issue is that efiSetPadMode is being called again even if brain_pin_markUnused wasn't called, which causes a Configuration Error. The call to efiSetPadMode is inside an #ifndef EFI_UNIT_TEST, so it doesn't run in unit tests.
Is there any way around this?

rusefillc, 2020-09-21:
@chuckwagoncomputing 

```
void gpio_pin_markUnused(ioportid_t port, ioportmask_t pin) {
	int index = getBrainIndex(port, pin);

	if (getBrainUsedPin(index) != NULL)
		pinRepository.totalPinsUsed--;
	getBrainUsedPin(index) = nullptr;
}```

we need unit-test implementation for getBrainIndex - that could be the beginning of unit test HW layer