rusefi/rusefi #1817: unit test for launch control

rusefillc, 2020-09-22:
we really need unit tests for launch control @shadowm60

shadowm60, 2020-09-23:
i understand, it is in work

rusefillc, 2020-11-19:
@shadowm60 poke

can I help to kick start you with this?

shadowm60, 2020-11-19:
Yes i do need some help.

Basically how this function is working, it has a periodikTimer task that checks the activation conditions. I would need to somehow run this from unit_tests to check enable/disable conditions. I did not managed to run this from unit_test.
In the meanwhile i found a bug while testing it in the car, and i do not want to fix it if it is not urgent, and no unit tests available.

@mck1117 sad that i should check the timing calculation only, but the whole logic/important stuff is in the periodic fast timer.
I just realized i could move this in a function that it is called from periodic task, then just call this.
Let me check it today and we can discuss about it on slack.

mck1117, 2020-11-19:
that's not quite what I said - I said you should try to refactor so that the PeriodicTask doesn't just have the whole implementation plopped in one function.

Here's an example of what I mean: https://github.com/rusefi/rusefi/pull/1967

This is much easier to write tests for, since you can test each piece individually, then mock out that that functionality to test whoever consumes it.

mck1117, 2020-11-19:
@shadowm60 if you want help with google mocks hit me up on slack, I'm happy to help walk you through it if that'd be helpful

shadowm60, 2020-11-19:
Ok, miss understood it then, i'll look into it and then if i still need help, we can talk, thanks

rusefillc, 2020-11-20:
See #1972

https://rusefi.com/docs/unit_tests_coverage/firmware/controllers/algo/launch_control.cpp.gcov.html

shadowm60, 2020-12-13:
I think this can be closed. We have 98% test covarage

closed 2020-12-13 by rusefillc