rusefi/rusefi #1819: cypress into GHA

rusefillc, 2020-09-26:
#1790 seems to compile in master now!

next step would be including it into GHA CI

rusefillc, 2020-09-27:
We have progress!

Now we need to fix the source code to make it compile on linux

the drama is about  #include "UsbDevice.h" vs usbdevice.h real file name 

funny think that #include "usbdevice.h" is also present!

@andreika-git need your help - what file name is right? which of imports is wrong?

andreika-git, 2020-09-27:
https://github.com/rusefi/rusefi/pull/1821

andreika-git, 2020-09-27:
Maybe we need to rename build_cypress -> build?

andreika-git, 2020-09-27:
It seems that renaming didn't help.. Still cannot find build/rusefi.hex and failing? 
Can you please merge #1821 and check what's going on?

closed 2020-09-27 by rusefillc