rusefi/rusefi #1848: Fail explicitly if clang is detected

rusefillc, 2020-10-03:
simulator and unit_tests should explicitly fail unless compiled with GCC - Apple use-case?

```
Xcode
XCode command line tools

gcc
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin14.5.0
Thread model: posix

g++
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin14.5.0
Thread model: posix

libtool
```

ruckusman, 2020-10-03:
printenv
TERM_PROGRAM=Apple_Terminal
SHELL=/bin/bash
TERM=xterm-256color
TMPDIR=/var/folders/8z/vc69wr5s7mx6jt5cn53znm2m0000gn/T/
Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.8tQhtChBdB/Render
TERM_PROGRAM_VERSION=343.7
TERM_SESSION_ID=5B09F0DC-C2CD-4CB3-AC4F-0231330871BC
USER=#########
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.ZNFj89kkq2/Listeners
__CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0
PATH=/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/Android
PWD=/Users/#########
LANG=en_AU.UTF-8
XPC_FLAGS=0x0
XPC_SERVICE_NAME=0
SHLVL=1
HOME=/Users/########
LOGNAME=########

Some variables
$PATH
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/Android

I think this is down to my build environment

going to add export CC=/usr/local/bin/gcc-10 to the .bash_profile

/usr/local/bin/gcc-10 --version
gcc-10 (Homebrew GCC 10.2.0) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I also just did a quick and dirty hack - making a second symlink in /usr/local/bin/gcc-10 called just gcc
Compilation goes further - the clang errors are still there, but as warnings, then it eventually fails on the same error

`make
No CCACHE_DIR
OS is []
Compiling chcore.c
clang: warning: argument unused during compilation: '-alms=build/lst/chcore.lst'
Compiling chsys.c
clang: warning: argument unused during compilation: '-alms=build/lst/chsys.lst'
Compiling chdebug.c
clang: warning: argument unused during compilation: '-alms=build/lst/chdebug.lst'
Compiling chtrace.c
clang: warning: argument unused during compilation: '-alms=build/lst/chtrace.lst'
Compiling chvt.c
clang: warning: argument unused during compilation: '-alms=build/lst/chvt.lst'
Compiling chschd.c
clang: warning: argument unused during compilation: '-alms=build/lst/chschd.lst'
Compiling chthreads.c
clang: warning: argument unused during compilation: '-alms=build/lst/chthreads.lst'
Compiling chregistry.c
clang: warning: argument unused during compilation: '-alms=build/lst/chregistry.lst'
Compiling chsem.c
clang: warning: argument unused during compilation: '-alms=build/lst/chsem.lst'
Compiling chmtx.c
clang: warning: argument unused during compilation: '-alms=build/lst/chmtx.lst'
Compiling chcond.c
clang: warning: argument unused during compilation: '-alms=build/lst/chcond.lst'
Compiling chevents.c
clang: warning: argument unused during compilation: '-alms=build/lst/chevents.lst'
Compiling chmsg.c
clang: warning: argument unused during compilation: '-alms=build/lst/chmsg.lst'
Compiling chdynamic.c
clang: warning: argument unused during compilation: '-alms=build/lst/chdynamic.lst'
Compiling chmboxes.c
clang: warning: argument unused during compilation: '-alms=build/lst/chmboxes.lst'
[... 39 lines cut ...]
../firmware/controllers/algo/nmea.c:253:10: warning: 'timp.tm_year' may be used uninitialized in this function [-Wmaybe-uninitialized]
  253 |  if (timp.tm_year > 0 ) {    // check if date field is valid
      |      ~~~~^~~~~~~~
clang: warning: argument unused during compilation: '-alms=build/lst/nmea.lst'
Compiling board.c
clang: warning: argument unused during compilation: '-alms=build/lst/board.lst'
Compiling memstreams.c
clang: warning: argument unused during compilation: '-alms=build/lst/memstreams.lst'
Compiling chprintf.c
clang: warning: argument unused during compilation: '-alms=build/lst/chprintf.lst'
Compiling crc.c
clang: warning: argument unused during compilation: '-alms=build/lst/crc.lst'
Compiling os_util.c
clang: warning: argument unused during compilation: '-alms=build/lst/os_util.lst'
Compiling histogram.c
clang: warning: argument unused during compilation: '-alms=build/lst/histogram.lst'
Compiling main.c
main.c: In function 'termination_handler':
main.c:59:43: warning: unused parameter 'id' [-Wunused-parameter]
   59 | static void termination_handler(eventid_t id) {
      |                                 ~~~~~~~~~~^~
clang: warning: argument unused during compilation: '-alms=build/lst/main.lst'
Compiling cyclic_buffer.cpp
clang: error: unsupported argument '-alms=build/lst/cyclic_buffer.lst' to option 'Wa,'
make: *** [build/obj/cyclic_buffer.o] Error 1`

ruckusman, 2020-10-03:
OK so I modified the build environment variables in the bash profile

I also made aliases in the /usr/local/bin directory for each of the 4 GCC components - I tried other variants of the aliases in the .bash_profile before making the symbolic links

ln -sf /usr/local/bin/c++-10 /usr/local/bin/c++
ln -sf /usr/local/bin/cpp-10 /usr/local/bin/cpp
ln -sf /usr/local/bin/g++-10 /usr/local/bin/g++
ln -sf /usr/local/bin/gcc-10 /usr/local/bin/gcc

`bash_profile

export PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin/sbin:/Applications/Android:$PATH

#export PATH=/usr/local/Cellar/gcc:$PATH

#export PATH:$PATH

export CC=/usr/local/bin/gcc

export CXX=/usr/local/bin/g++

export CPP=/usr/local/bin/cpp

export LD=/usr/local/bin/gcc

alias c++=/usr/local/bin/c++

alias g++=/usr/local/bin/g++

alias gcc=/usr/local/bin/gcc

alias cpp=/usr/local/bin/cpp

alias ld=/usr/local/bin/gcc

alias cc=/usr/local/bin/gcc`

printenv 

`TERM_PROGRAM=Apple_Terminal
SHELL=/bin/bash
TERM=xterm-256color
TMPDIR=/var/folders/8z/vc69wr5s7mx6jt5cn53znm2m0000gn/T/
Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.ah8o816ddl/Render
TERM_PROGRAM_VERSION=343.7
TERM_SESSION_ID=0D4C8F62-C0F4-47A8-90F7-9ABEB0252284
USER=########
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.2pxZ8CdCFD/Listeners
__CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0
CPP=/usr/local/bin/cpp
PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin/sbin:/Applications/Android:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
LD=/usr/local/bin/gcc
PWD=/Users/########
LANG=en_AU.UTF-8
XPC_FLAGS=0x0
CXX=/usr/local/bin/g++
XPC_SERVICE_NAME=0
SHLVL=1
HOME=/Users/########
LOGNAME=########
CC=/usr/local/bin/gcc`

The clang warning still occurs throughout the compilation related to each module, but the build continues
clang: warning: argument unused during compilation: '-alms=build/lst/boards.lst'

There is one final error, seems to be to do with library linking at the final step
 
`Linking build/rusefi_simulator
ld: unknown option: -Map=build/rusefi_simulator.map
collect2: error: ld returned 1 exit status
make: *** [build/rusefi_simulator] Error 1`

rusefillc, 2020-10-03:
this is NOT a ticket to make this work with clang

this is a ticket to FAIL with clang

ruckusman, 2020-10-03:
Yes understood - I've finally managed to get rid of clang altogether and it's using the GCC

mck1117, 2020-10-03:
I don't remember if we do on other platforms, but you might want to link with g++ instead of gcc

ruckusman, 2020-10-04:
The aliases which I've set in the .bash_profile don't seem to take even though they show up in printenv - the symbolic links to overcome the -10 version extensions work, however there isn't a binutils version available via brew so the OSX ld is always called at the final step

I've checked and although I've installed brew binutils it isn't there - so the compile fails at the final linking step

Output of final lines from make -d

Finished prerequisites of target file `build/rusefi_simulator'.
  Must remake target `build/rusefi_simulator'.
Putting child 0x7fbca1cecac0 (build/rusefi_simulator) PID 15792 on the chain.
Live child 0x7fbca1cecac0 (build/rusefi_simulator) PID 15792 
Linking build/rusefi_simulator
Reaping winning child 0x7fbca1cecac0 PID 15792 
Live child 0x7fbca1cecac0 (build/rusefi_simulator) PID 15793 
ld: unknown option: -Map=build/rusefi_simulator.map
collect2: error: ld returned 1 exit status
Reaping losing child 0x7fbca1cecac0 PID 15793 
make: *** [build/rusefi_simulator] Error 1
Removing child 0x7fbca1cecac0 PID 15793 from chain.

I'd say this is a bust which means I need to get my act together and do a complete instal of Linux on the other Macbook Pro

I did find this, which is interesting, as I've noticed that the OS is always blank 'OS is []'

https://github.com/netinfox/ChibiOS_OSX

I'm going to see if I can select a few of the options from within the Makefile to see if they make any difference 

Happy to keep trying with this if you have any suggestions and it means it doesn't break the compilation on other machines in the process

ruckusman, 2020-10-04:
> I don't remember if we do on other platforms, but you might want to link with g++ instead of gcc

I tried changing the symbolic link to g++ and it falls over very quickly

make

> No CCACHE_DIR
OS is []
Compiling chcore.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
clang: warning: argument unused during compilation: '-alms=build/lst/chcore.lst'
Compiling chsys.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
clang: warning: argument unused during compilation: '-alms=build/lst/chsys.lst'
Compiling chdebug.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
clang: warning: argument unused during compilation: '-alms=build/lst/chdebug.lst'
Compiling chtrace.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
clang: warning: argument unused during compilation: '-alms=build/lst/chtrace.lst'
Compiling chvt.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
clang: warning: argument unused during compilation: '-alms=build/lst/chvt.lst'
Compiling chschd.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
clang: warning: argument unused during compilation: '-alms=build/lst/chschd.lst'
Compiling chthreads.c
cc1plus: warning: command-line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
cc1plus: warning: command-line option '-std=gnu99' is valid for C/ObjC but not for C++
In file included from ../firmware/ChibiOS/os/rt/include/ch.h:110,
                 from ../firmware/ChibiOS/os/rt/src/chthreads.c:51:
../firmware/ChibiOS/os/rt/src/chthreads.c: In function 'thread_t* chThdCreateSuspendedI(const thread_descriptor_t*)':
../firmware/ChibiOS/os/common/ports/SIMIA32/chcore.h:200:7: error: invalid conversion from 'void*' to 'uint8_t*' {aka 'unsigned char*'} [-fpermissive]
  200 |   p = (void *)((((uint32_t)(p) - (uint32_t)(mod)) & ~(uint32_t)(mask)) + (uint32_t)(mod)) \
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       |
      |       void*
../firmware/ChibiOS/os/common/ports/SIMIA32/chcore.h:212:3: note: in expansion of macro 'AALIGN'
  212 |   AALIGN(esp, 15, 8);                                                       \
      |   ^~~~~~
../firmware/ChibiOS/os/rt/src/chthreads.c:190:3: note: in expansion of macro 'PORT_SETUP_CONTEXT'
  190 |   PORT_SETUP_CONTEXT(tp, tdp->wbase, tp, tdp->funcp, tdp->arg);
      |   ^~~~~~~~~~~~~~~~~~
In file included from ../firmware/ChibiOS/os/rt/include/ch.h:107,
                 from ../firmware/ChibiOS/os/rt/src/chthreads.c:51:
../firmware/ChibiOS/os/rt/src/chthreads.c: In function 'thread_t* chThdCreateStatic(void*, size_t, tprio_t, tfunc_t, void*)':
../firmware/ChibiOS/os/rt/src/chthreads.c:335:44: error: invalid conversion from 'void*' to 'stkalign_t*' [-fpermissive]
  335 |   chDbgAssert(chRegFindThreadByWorkingArea(wsp) == NULL,
      |                                            ^~~
      |                                            |
      |                                            void*
../firmware/ChibiOS/os/rt/include/chdebug.h:131:11: note: in definition of macro 'chDbgAssert'
  131 |     if (!(c)) {                                                             \
      |           ^
In file included from ../firmware/ChibiOS/os/rt/include/ch.h:120,
                 from ../firmware/ChibiOS/os/rt/src/chthreads.c:51:
../firmware/ChibiOS/os/rt/include/chregistry.h:113:54: note:   initializing argument 1 of 'thread_t* chRegFindThreadByWorkingArea(stkalign_t*)'
  113 |   thread_t *chRegFindThreadByWorkingArea(stkalign_t *wa);
      |                                          ~~~~~~~~~~~~^~
In file included from ../firmware/ChibiOS/os/rt/include/ch.h:110,
                 from ../firmware/ChibiOS/os/rt/src/chthreads.c:51:
../firmware/ChibiOS/os/common/ports/SIMIA32/chcore.h:200:7: error: invalid conversion from 'void*' to 'uint8_t*' {aka 'unsigned char*'} [-fpermissive]
  200 |   p = (void *)((((uint32_t)(p) - (uint32_t)(mod)) & ~(uint32_t)(mask)) + (uint32_t)(mod)) \
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       |
      |       void*
../firmware/ChibiOS/os/common/ports/SIMIA32/chcore.h:212:3: note: in expansion of macro 'AALIGN'
  212 |   AALIGN(esp, 15, 8);                                                       \
      |   ^~~~~~
../firmware/ChibiOS/os/rt/src/chthreads.c:359:3: note: in expansion of macro 'PORT_SETUP_CONTEXT'
  359 |   PORT_SETUP_CONTEXT(tp, wsp, tp, pf, arg);
      |   ^~~~~~~~~~~~~~~~~~
../firmware/ChibiOS/os/rt/src/chthreads.c: In function 'void chThdRelease(thread_t*)':
../firmware/ChibiOS/os/rt/src/chthreads.c:449:22: error: invalid conversion from 'void*' to 'memory_pool_t*' [-fpermissive]
[... cut ...]
      |                  ~~~~^~~~~
      |                      |
      |                      void*
In file included from ../firmware/ChibiOS/os/rt/include/ch.h:132,
                 from ../firmware/ChibiOS/os/rt/src/chthreads.c:51:
../firmware/ChibiOS/os/common/oslib/include/chmempools.h:160:34: note:   initializing argument 1 of 'void chPoolFree(memory_pool_t*, void*)'
  160 |   void chPoolFree(memory_pool_t *mp, void *objp);
      |                   ~~~~~~~~~~~~~~~^~
make: *** [build/obj/chthreads.o] Error 1

mck1117, 2020-10-04:
> I tried changing the symbolic link to g++ and it falls over very quickly

I don't mean symbolic link, I mean the linker command (`ld`).

This part:
`alias ld=/usr/local/bin/gcc`
change it to
`alias ld=/usr/local/bin/g++`

ruckusman, 2020-10-04:
I've tried three variants of those aliases 
alias ld=/usr/local/bin/g++
alias ld='/usr/local/bin/g++'
alias ld="/usr/local/bin/g++"

printenv doesn't reflect any of them, in addition `which ld` yields 

> /usr/bin/ld

So I don't think that environment variable in the .bash_profile is taking effect

ruckusman, 2020-10-12:
Just a update on this - I installed Linux MX19 on the Macbook which is no easy task with the hybrid graphics
Got it all compiled and working under Linux including the simulator which was the initial goal.
I will do a full write-up on the forum to help anyone else considering Linux in the coming days.

On Mac OSX:
I did manage to get all of the libraries compiled using both GCC and CC (clang). on a later model Macbook with High Sierra installed.
The GCC compiled libraries wouldn't link with the Mac OSX ld and the GCC ld isn't available on Mac OSX via homebrew, so I left that task to go the Linux route, but kept copious notes, so will return to it.

As I've kept track of all of the steps I should be able to reconstruct the small modification of the *.mk files to complete the compilation steps prior to linking.
There weren't actually that many modifications necessary, and I was up to correcting some linker options

rusefillc, 2020-10-12:
@ruckusman do you think you already have changes to pull request?

ruckusman, 2020-10-12:
You'll have to give me instructions on how to do that, or would you prefer if I wait until I have successfully compiled and linked to a finished simulator binary?

rusefillc, 2020-12-17:
Change of plans we now support clang?! #2086

closed 2020-12-17 by rusefillc