rusefi/rusefi #1849: HW CI has failed 11 times since Oct 2

rusefillc, 2020-10-03:
```
 [java] I 201003 121932.957 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 121932.957 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 121932.958 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 640 byte(s) arrived
     [java] I 201003 121932.958 [communication executor1] IoStream - [serial] composite log body: waiting for 835 byte(s)
     [java] I 201003 121932.959 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 197 byte(s) arrived
     [java] I 201003 121932.959 [communication executor1] CompositeParser - Got composite length=831
     [java] I 201003 121932.960 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 121932.961 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 166 byte(s) arrived
     [java] I 201003 121932.961 [communication executor1] IoStream - [serial] text body: waiting for 164 byte(s)
     [java] I 201003 121932.961 [communication executor1] LinkManager - msg,confirmation_rpm 900:7,msg,Emulating position sensor(s). RPM=900,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,
     [java] I 201003 121932.961 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_rpm 900:7
     [java] I 201003 121932.961 [communication executor1] MessagesCentral - postMessage EngineState: Emulating position sensor(s). RPM=900
     [java] I 201003 121932.961 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121932.961 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.057 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 121933.058 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 295 byte(s) arrived
     [java] I 201003 121933.058 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 121933.059 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 121933.059 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 292 byte(s) arrived
     [java] I 201003 121933.059 [communication executor1] IoStream - [serial] composite log body: waiting for 290 byte(s)
     [java] I 201003 121933.059 [communication executor1] CompositeParser - Got composite length=286
     [java] I 201003 121933.060 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 121933.061 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 97 byte(s) arrived
     [java] I 201003 121933.061 [communication executor1] IoStream - [serial] text body: waiting for 95 byte(s)
     [java] I 201003 121933.061 [communication executor1] LinkManager - msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,
     [java] I 201003 121933.061 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.061 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.158 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 121933.158 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 295 byte(s) arrived
     [java] I 201003 121933.159 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 121933.159 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 121933.161 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 1600 byte(s) arrived
     [java] I 201003 121933.161 [communication executor1] IoStream - [serial] composite log body: waiting for 1600 byte(s)
     [java] I 201003 121933.161 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 2 byte(s) arrived
     [java] I 201003 121933.161 [communication executor1] CompositeParser - Got composite length=1596
     [java] I 201003 121933.163 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 121933.163 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 97 byte(s) arrived
     [java] I 201003 121933.163 [communication executor1] IoStream - [serial] text body: waiting for 95 byte(s)
     [java] I 201003 121933.163 [communication executor1] LinkManager - msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,
     [java] I 201003 121933.163 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.163 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.258 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 121933.259 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 295 byte(s) arrived
     [java] I 201003 121933.259 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 121933.259 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 121933.260 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 640 byte(s) arrived
     [java] I 201003 121933.260 [communication executor1] IoStream - [serial] composite log body: waiting for 910 byte(s)
     [java] I 201003 121933.261 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 272 byte(s) arrived
     [java] I 201003 121933.261 [communication executor1] CompositeParser - Got composite length=906
     [java] I 201003 121933.262 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 121933.262 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 97 byte(s) arrived
     [java] I 201003 121933.262 [communication executor1] IoStream - [serial] text body: waiting for 95 byte(s)
     [java] I 201003 121933.263 [communication executor1] LinkManager - msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,
     [java] I 201003 121933.263 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.263 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 121933.358 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 121933.359 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 295 byte(s) arrived
     [java] I 201003 121933.359 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 121933.359 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 121933.360 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 960 byte(s) arrived
     [java] I 201003 121933.360 [communication executor1] IoStream - [serial] composite log body: waiting for 990 byte(s)
     [java] I 201003 121933.362 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 32 byte(s) arrived
     [java] I 201003 121933.362 [communication executor1] CompositeParser - Got composite length=986
     [java] I 201003 121933.364 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 121933.369 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 3840 byte(s) arrived
     [java] I 201003 121933.369 [communication executor1] IoStream - [serial] text body: waiting for 4127 byte(s)
     [java] I 201003 121933.369 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 289 byte(s) arrived
     [java] I 201003 121933.370 [communication executor1] LinkManager - msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,wave_chart,t1!d_21!0!t1!u_22!56!t1!d_23!111!c1!d!166!t1!u_24!168!t1!d_25!222!t1!u_26!279!t1!d_27!333!t1!u_28!389!t1!d_29!444!i3!d!448!r!900!455!t1!u_30!500!t1!d_31!555!t1!u_32!611!t1!d_33!666!t1!u_34!722!map!u!749!t1!d_35!778!t1!u_36!834!t1!d_37!888!t1!u_38!945!t1!d_39!999!t1!u_40!1056! [... cut ...]
     [java]     at org.apache.tools.ant.Project.executeTarget(Project.java:1364)
     [java]     at org.apache.tools.ant.helper.DefaultExecutor.executeTargets(DefaultExecutor.java:41)
     [java]     at org.apache.tools.ant.Project.executeTargets(Project.java:1254)
     [java]     at org.apache.tools.ant.Main.runBuild(Main.java:830)
     [java]     at org.apache.tools.ant.Main.startAnt(Main.java:223)
     [java]     at org.apache.tools.ant.launch.Launcher.run(Launcher.java:284)
     [java]     at org.apache.tools.ant.launch.Launcher.main(Launcher.java:101)

```

rusefillc, 2020-10-03:
```\
java] I 201003 122612.313 [communication executor1] IoStream - [serial] text body: waiting for 5 byte(s)
     [java] I 201003 122612.413 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 122612.414 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 295 byte(s) arrived
     [java] I 201003 122612.414 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 122612.414 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 122612.414 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 3 byte(s) arrived
     [java] I 201003 122612.414 [communication executor1] IoStream - [serial] composite log body: waiting for 615 byte(s)
     [java] I 201003 122612.415 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 320 byte(s) arrived
     [java] I 201003 122612.416 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 294 byte(s) arrived
     [java] I 201003 122612.416 [communication executor1] CompositeParser - Got composite length=611
     [java] I 201003 122612.417 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 122612.417 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 7 byte(s) arrived
     [java] I 201003 122612.417 [communication executor1] IoStream - [serial] text body: waiting for 5 byte(s)
     [java] I 201003 122612.518 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 122612.519 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 295 byte(s) arrived
     [java] I 201003 122612.519 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201003 122612.520 [communication executor1] IoStream - composite log header: waiting for 2 byte(s)
     [java] I 201003 122612.520 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 142 byte(s) arrived
     [java] I 201003 122612.520 [communication executor1] IoStream - [serial] composite log body: waiting for 140 byte(s)
     [java] I 201003 122612.520 [communication executor1] CompositeParser - Got composite length=136
     [java] I 201003 122612.521 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201003 122612.526 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 4096 byte(s) arrived
     [java] I 201003 122612.526 [communication executor1] IoStream - [serial] text body: waiting for 4393 byte(s)
     [java] I 201003 122612.527 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 64 byte(s) arrived
     [java] I 201003 122612.527 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 235 byte(s) arrived
     [java] I 201003 122612.527 [communication executor1] LinkManager - msg,Engine runs we can disengage the starter,msg,isCylinderCleanupMode No,msg,Engine runs we can disengage the starter,wave_chart,t1!d_119!0!t1!u_120!83!t1!d_121!166!t1!u_122!250!t1!d_123!333!t1!u_124!417!t1!d_125!499!t1!u_126!583!t1!d_127!666!t1!u_128!750!t1!d_129!833!t1!u_130!917!t1!d_131!999!t1!u_132!1083!t1!d_133!1166!t1!u_134!1250!t1!d_135!1333!t1!u_136!1417!t1!d_137!1499!t1!u_138!1583!t1!d_139!1666!t1!u_140!1750!t1!d_141!1833!t1!u_142!1917!t1!d_143!1999!t1!u_144!2083!t1!d_145!2166!t1!u_146!2250!t1!d_147!2333!t1!u_148!2417!t1!d_149!2499!t1!u_150!2583!t1!d_151!2666!t1!u_152!2750!t1!d_153!2833!t1!u_154!2917!t1!d_155!2999!t1!u_156!3083!t1!d_157!3166!t1!u_158!3250!t1!d_159!3333!t1!u_160!3417!t1!d_161!3499!t1!u_162!3583!t1!d_163!3666!t1!u_164!3750!t1!d_165!3833!t1!u_166!3917!t1!d_167!3999!t1!u_168!4083!t1!d_169!4166!t1!u_170!4250!t1!d_171!4333!t1!u_172!4417!t1!d_173!4499!t1!u_174!4583!t1!d_175!4666!t1!u_176!4750!t1!d_177!4833!t1!u_178!4917!t1!d_179!4999!t1!u_180!5083!t1!d_181!5166!t1!u_182!5250!t1!d_183!5333!t1!u_184!5417!t1!d_185!5499!t1!u_186!5583!t1!d_187!5666!t1!u_188!5750!t1!d_189!5833!t1!u_190!5917!t1!d_191!5999!t1!u_192!6084!t1!d_193!6166!t1!u_194!6250!t1!d_195!6333!t1!u_0!6584!t1!d_1!6833!t1!u_2!6917!t1!d_3!6999!t1!u_4!7083!t1!d_5!7166!t1!u_6!7250!t1!d_7!7333!t1!u_8!7417!t1!d_9!7499!t1!u_10!7583!t1!d_11!7666!t1!u_12!7750!t1!d_13!7833!t1!u_14!7917!t1!d_15!7999!t1!u_16!8083!t1!d_17!8166!t1!u_18!8250!t1!d_19!8333!t1!u_20!8417!t1!d_21!8499!t1!u_22!8583!t1!d_23!8666!t1!u_24!8750!t1!d_25!8833!t1!u_26!8917!t1!d_27!8999!t1!u_28!9083!t1!d_29!9166!t1!u_30!9250!t1!d_31!9333!t1!u_32!9417!t1!d_33!9499!t1!u_34!9583!t1!d_35!9666!t1!u_36!9750!t1!d_37!9833!t1!u_38!9917!t1!d_39!9999!t1!u_40!10083!t1!d_41!10166!t1!u_42!10250!t1!d_43!10333!t1!u_44!10417!t1!d_45!10499!t1!u_46!10583!t1!d_47!10666!t1!u_48!10750!t1!d_49!10833!t1!u_50!10917!t1!d_51!10999!t1!u_52!11084!t1!d_53!11166!t1!u_54!11250!t1!d_55!11333!t1!u_56!11417!t1!d_57!11499!t1!u_58!11583!t1!d_59!11666!t1!u_60!11750!t1!d_61!11833!t1!u_62!11917!t1!d_63!11999!t1!u_64!12083!t1!d_65!12166!t1!u_66!12250!t1!d_67!12333!t1!u_68!12417!t1!d_69!12499!t1!u_70!12583!t1!d_71!12666!t1!u_72!12750!t1!d_73!12833!t1!u_74!12917!t1!d_75!12999!t1!u_76!13083!t1!d_77!13166!t1!u_78!13250!t1!d_79!13333!t1!u_80!13417!t1!d_81!13499!t1!u_82!13583!t1!d_83!13666!t1!u_84!13750!t1!d_85!13833!t1!u_86!13917!t1!d_87!13999!t1!u_88!14083!t1!d_89!14166!t1!u_90!14250!t1!d_91!14333!t1!u_92!14417!t1!d_93!14499!t1!u_94!14583!t1!d_95!14666!t1!u_96!14750!t1!d_97!14833!t1!u_98!14917!t1!d_99!14999!t1!u_100!15083!t1!d_101!15166!t1!u_102!15250!t1!d_103!15333!t1!u_104!15417!t1!d_105!15499!t1!u_10 [... cut ...]
     [java]     at org.apache.tools.ant.Project.executeTargets(Project.java:1254)
     [java]     at org.apache.tools.ant.Main.runBuild(Main.java:830)
     [java]     at org.apache.tools.ant.Main.startAnt(Main.java:223)
     [java]     at org.apache.tools.ant.launch.Launcher.run(Launcher.java:284)
     [java]     at org.apache.tools.ant.launch.Launcher.main(Launcher.java:101)

BUILD FAILED
<http://127.0.0.1:8080/job/integration_and_primary_bundle/ws/java_console/build.xml>:185: Java returned: -1
```

rusefillc, 2020-10-04:
still

```
[java] I 201003 200545.627 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 320 byte(s) arrived
     [java] I 201003 200545.639 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 64 byte(s) arrived
     [java] I 201003 200545.692 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 320 byte(s) arrived
     [java] I 201003 200545.807 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 205 byte(s) arrived
     [java] I 201003 200545.807 [communication executor1] LinkManager - msg,confirmation_rpm 20000:9,msg,Emulating position sensor(s). RPM=20000,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,calc 0: action fan value 0.00,msg,calc 1: action CLT value 0.00,msg,calc 2: action cfg_fanOffTemperature value 91.00,msg,calc 3: action 3 value 0.00,msg,calc 4: action 6 value 0.00,msg,calc 5: action CLT value 0.00,msg,calc 6: action cfg_fanOnTemperature value 95.00,msg,calc 7: action 3 value 0.00,msg,calc 8: action 7 value 0.00,msg,calc 9: action 119 value 1.00,msg,calc 10: action 7 value 1.00,msg,setPin fan on,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,msg,Engine runs we can disengage the starter,
     [java] I 201003 200545.807 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_rpm 20000:9
     [java] I 201003 200545.808 [communication executor1] MessagesCentral - postMessage EngineState: Emulating position sensor(s). RPM=20000
     [java] I 201003 200545.808 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.808 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.808 [communication executor1] MessagesCentral - postMessage EngineState: calc 0: action fan value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 1: action CLT value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 2: action cfg_fanOffTemperature value 91.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 3: action 3 value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 4: action 6 value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 5: action CLT value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 6: action cfg_fanOnTemperature value 95.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 7: action 3 value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 8: action 7 value 0.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 9: action 119 value 1.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: calc 10: action 7 value 1.00
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: setPin fan on
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.809 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] MessagesCentral - postMessage EngineState: Engine runs we can disengage the starter
     [java] I 201003 200545.810 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
     [java] I 201003 200546.431 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 291 byte(s) arrived
     [java] I 201003 200546.431 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
[... cut ...]
     [java]     at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
     [java]     at sun.reflect.GeneratedMethodAccessor9.invoke(Unknown Source)
     [java]     at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
     [java]     at java.lang.reflect.Method.invoke(Method.java:498)
     [java]     at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:99)
     [java]     at org.apache.tools.ant.Task.perform(Task.java:350)
```

rusefillc, 2020-10-04:
last successful build

[image]

rusefillc, 2020-10-04:
First unsuccessful build

@mck1117 

[image]

rusefillc, 2020-10-04:
@mck1117 looks like  #1831 has introduced the issue

rusefillc, 2020-10-04:
this incident reminds me of https://github.com/rusefi/rusefi_documentation/issues/28 :(

rusefillc, 2020-10-04:
See also #1850

rusefillc, 2020-10-05:
#1841 re-applied

Overall status: with #1831 reverted we are back in the green.

I assume we are still interested in the #1831 changes so next step figuring out what went wrong?

mck1117, 2020-10-05:
current performance status:  trigger stimulator performance is improved a bit, between trigger decode and PWM performance.  Interrupt total time dropped from 45 to 38 us (-15%).  Some of that was PWM improvement, and some was speeding up the trigger decode itself (22us -> 19us, -14%)

rusefillc, 2020-10-05:
And HW CI is back at 20K https://github.com/rusefi/rusefi/commit/2398741ed3f419d4da43567830b8b8928e8268f8

rusefillc, 2020-10-07:
HW CI has never been as reliable as these days

[image]

rusefillc, 2020-10-07:
the random lsize fatal is not happening for whatever reason #884 could be luck could be related to performance fixes

rusefillc, 2020-10-16:
#888 failure with #749282c80717d8f1fd7ea7d7d1aa22f660b54a1f

rusefillc, 2020-10-18:
With 16k and handling only rise event we get stability

[image]

mck1117, 2020-10-26:
I guess this is fixed with some perf improvements, and some changes to exactly what we test?

closed 2020-10-26 by rusefillc