rusefi/rusefi #1852: poor-mans-profiler

rusefillc, 2020-10-04:
would be great to learn how to sample stack trace repetatively

https://interrupt.memfault.com/blog/profiling-firmware-on-cortex-m#poor-mans-profiler

rusefillc, 2020-10-04:
I can invoke

arm-none-eabi-gdb -ex "set pagination off" -ex "target remote :3333" -ex "thread apply all bt"  -ex "thread apply all bt" -ex "thread apply all bt" -batch build/rusefi.elf

but that always prints same stack trace for whatever reason

rusefillc, 2020-10-04:
Actually both legacy atollic truestudio and current stm32cubeIde have build-in sampling profiling for free?

I was not able to get this working with stm32cube

but I got this working with Atollic_TrueSTUDIO_for_STM32_windows_x86_v9.3.0_20190212-0734.exe - adding a magic file into .settings folder was everything needed

https://youtu.be/J8Blmf-cWKM

rusefillc, 2020-10-04:
[image]

rusefillc, 2020-10-04:
Something is not right :( I am only getting one batch of initial data regardless of how long sampling runs for

rusefillc, 2020-10-04:
PB3 pin conflict was one of the problems

PB3 is both SPI and SWO wire

mck1117, 2020-11-26:
do we care?

rusefillc, 2020-11-26:
I believe yes, we care - sampling adds value to the profiling we have