rusefi/rusefi #1873: Trigger sync lost every 2^32 CPU cycles

mck1117, 2020-10-11:
Something is wrong with hardware trigger inputs - I'm losing sync exactly every 2^32 cpu cycles, or 25.59 seconds.

[image]

[image]

mck1117, 2020-10-11:
CLT, IAT, TPS also drop out just before the trigger error

rusefillc, 2020-10-11:
I believe I have three boards WITH this issue and one board WITHOUT

I do not have definitive data but I have a feeling that the affected boards are stm32f427 while the happy board is stm32f407

rusefillc, 2020-10-11:
https://github.com/rusefi/rusefi/releases/tag/20200831 not affected?

mck1117, 2020-10-11:
Here's a hint:

[image]

rusefillc, 2020-10-11:
caused by #1858

commit 018db0df (2020-10-12): Fix #1873 - compute dynamic PWM iterationLimit (#1874)

* fix

* remove define

* add an assert

* typo

* don't do a 64bit divide

* oh that's a float
closed 2020-10-12 by rusefillc