rusefi/rusefi #1900: Composite logger is not 100% functional

shadowm60, 2020-10-23:
while we can get some "snapshots" of the trigger, it is not working correct.

Hits from @960
"I'll share a secret with you that took me days to figure out.
The third bit in composite-logger where you have TDC is actually a flag to toggle the top graph.
It's called "Trigger", and is set to 1 when cam is active."

checking rusefi.ini i can find this:
	; these names are hard-coded inside TS
	recordField = priLevel,          "PriLevel",     0,          1,          1.0,    "Flag"
        recordField = secLevel,          "SecLevel",     1,          1,          1.0,    "Flag"
        recordField = trigger,           "Trigger",      2,          1,          1.0,    "Flag"
        recordField = sync,              "Sync",         3,          1,          1.0,    "Flag"
	recordField = time,              "Time",         8,         32,         0.01,    "ms"

But in the sources we have the following :
typedef struct __attribute__ ((packed)) {
	// the whole order of all packet bytes is reversed, not just the 'endian-swap' integers
	uint32_t timestamp;
	// unfortunately all these fields are required by TS...
	bool priLevel : 1;
	bool secLevel : 1;
	bool isTDC : 1;
	bool sync : 1;
	bool coil : 1;
	bool injector : 1;
} composite_logger_s;

What i understand from the code snippet shared by 960 is that when we log cam signal we have to put the trigger bit as true.

shadowm60, 2020-10-23:
Looks like time resolution is not right in the ini file, with this configuration i have the same pulse with as Console is displaying
		recordField = time,              "Time",         8,         32,        **0.001**,    "ms"

960, 2020-10-24:
[LoggerDefinition]
 

;  loggerDef = compositeLogger, "Primary Trigger Logger", tooth
  loggerDef = compositeLogger, "Trigger Logger", composite
        startCommand = "l\x01"
        stopCommand  = "l\x02"
        dataReadCommand = "L"
        dataReadTimeout = 50000 ; time in ms
        dataReadyCondition = { toothLogReady == 1}
        continuousRead = true
        ; each packet is @@COMPOSITE_PACKET_SIZE@@ and we have @@COMPOSITE_PACKET_COUNT@@ of those
        dataLength = @@COMPOSITE_DATA_LENGTH@@
		recordDef =   0,   0,  5

 ;recordField = Name, 		HeaderName, startBit, bitCount, scale, 	units, 	updateCondition
 
 	; Composite compatability fields
 	recordField =  	priLevel, 		"PriLevel", 	0, 		1, 		1.0, 	"H/L"; , { trigger == 0 }
 	recordField =  	secLevel, 		"SecLevel", 	1, 		1, 		1.0, 	"H/L"; , { trigger == 1 }
 	recordField =  	trigger, 		"Trigger", 		2, 		1, 		1.0, 	""
	recordField =  	sync, 			"Sync", 		3, 		1, 		1.0, 	""
	recordField = 	refTime, 		"RefTime", 		8, 		32, 	0.001, 	"uS"

	;recordField = toothTime,         "ToothTime",     0,          32,       1.0,    "uS"

       ; hidden calcField serves as intermediate variable to track an overflow
       calcField = rollOverCount, "RollCtr", "#", { isNaN(rollOverCount) ? 0 : refTime < pastValue(refTime, 1) ? rollOverCount+1 : rollOverCount }, hidden
       
       ;create a running time that doesn't overflow
       ;placed up here so it can be referenced by toothtime, hidden as it is redundant data
       calcField = noOverFlowtime, "No OverFlow Time", "ms", {refTime + (2657900 * rollOverCount)}, hidden

 	 calcField = toothTime, "ToothTime", "ms", { noOverFlowtime - pastValue(noOverFlowtime, 1) }
       ; currently the "Time" field needs to be at the end of the row for it to jump to the
       ; record when clicking on the chart.
       ; So this is created here
       calcField = time, "Time", "ms", { noOverFlowtime }

 ; currently the "Time" field needs to be at the end of the row for it to jump to the
 ; record when clicking on the chart.
 ; So this is created here
  loggerDef = tooth, "Tooth Logger", tooth
       ;dataReadCommand = "r\\x00\\xf4\\x00\\x00\\x04\\x00" ; standard TS command format
       startCommand = "H"
       stopCommand = "h"
       ;dataReadCommand = "T" ; Basic TS command format
       dataReadCommand = "L" ; Basic TS command format. Note that this is shared with the composite logger. Firmware detects which log is currently running
       dataReadTimeout = 50000 ; time in ms
       continuousRead = true
       dataReadyCondition = { toothLogReady == 1 }
       dataLength =  @@COMPOSITE_DATA_LENGTH@@; in bytes, including headers, footers and data (not used). 4 bytes * 127 entries
       ;dataLength = 128 ; in bytes, including headers, footers and data (not used)

       ;recordDef = headerLen. footerLen, recordLen
       recordDef =   0,   0,   4; in bytes, the recordLen is for each record, currently limited to 4 bytes

       ;recordField = Name,          HeaderName,      startBit,   bitCount,   scale,  units, updateCondition
       recordField = toothTime,         "ToothTime",     0,          32,       1.0,    "uS"

960, 2020-10-24:
bit of a mess :-)

shadowm60, 2020-10-24:
@960 can you give me more details what should the board replay/do on the H command?

960, 2020-10-24:
Its a simple tooth logger. It's not making any extra value, but I tought I needed the toothtime together with composite logger for it to work.

My whole mess of a file..
```

/*
 * @file tooth_logger.cpp
 *
 * @date Jul 7, 2019
 * @author Matthew Kennedy
 */

#include "tooth_logger.h"
#include "engine.h"
#include "global.h"

#if EFI_TOOTH_LOGGER

#include "tunerstudio_io.h"

#include <cstddef>
#include "efitime.h"
#include "efilib.h"
#include "tunerstudio_outputs.h"

EXTERN_ENGINE;

typedef struct __attribute__ ((packed)) {
    uint32_t timestamp;
} tooth_logger_s;

typedef struct __attribute__ ((packed)) {
	uint32_t timestamp;
	bool priLevel : 1;
	bool secLevel : 1;
	bool trigger : 1;
	bool sync : 1;
} composite_logger_s;

uint8_t compositeLogHistory[TOOTH_PACKET_COUNT];

static composite_logger_s buffer[COMPOSITE_PACKET_COUNT] CCM_OPTIONAL;
static tooth_logger_s toothHistory[COMPOSITE_PACKET_COUNT] CCM_OPTIONAL;

static size_t NextIdx = 0;
static volatile bool ToothLoggerEnabled = false;

static uint32_t lastEdgeTimestamp = 0;
static uint32_t nowUs = 0;
static bool trigger1;
static bool trigger2;
static bool camEdge;

int getCompositeRecordCount() {
	return NextIdx;
}

bool valueLogged = false;

bool validVvt = false;
bool validSync = false;

uint32_t lasttoothtime;
uint32_t lastcranktime;
uint32_t lastcamtime;
uint32_t thistoothtime;

static inline void addToothLogEntry(uint32_t lasttoothtime, bool whichtooth, bool edge) {
	if((engine->toothLogEnabled == true) || (engine->compositeLogEnabled == true))
	  {
	    bool valueLogged = false;
	if (engine->toothLogEnabled == true) {
		toothHistory[NextIdx].timestamp = SWAP_UINT32(lasttoothtime);
		valueLogged = true;
	}
	else if (engine->compositeLogEnabled == true) {

	nowUs = getTimeNowUs();
	buffer[NextIdx].timestamp = SWAP_UINT32(nowUs);
	if (whichtooth == trigger1) {
	buffer[NextIdx].priLevel = whichtooth;
	}
	if (whichtooth == trigger2) {
	buffer[NextIdx].secLevel = whichtooth;
	}
	buffer[NextIdx].trigger = edge;
	buffer[NextIdx].sync = engine->triggerCentral.triggerState.shaft_is_synchronized;

	uint32_t gap2 = getTimeNowUs() - thistoothtime;
	toothHistory[NextIdx].timestamp = SWAP_UINT32(gap2);
	thistoothtime = getTimeNowUs();
	valueLogged = true;
	}
	if (valueLogged == true) {
		NextIdx++;
	}
	static_assert(sizeof(composite_logger_s) == COMPOSITE_PACKET_SIZE, "composite packet size");

	if (NextIdx >= COMPOSITE_PACKET_COUNT - 1) {
		NextIdx = 0;
	}
}
}

void LogCrankTooth(trigger_value_e level DECLARE_ENGINE_PARAMETER_SUFFIX) {
	// bail if we aren't enabled
	if (!ToothLoggerEnabled) {
		return;
	}
	switch (level) {
	case TV_FALL:
		trigger1 = false;
		camEdge = false;
		break;
	case TV_RISE:
		trigger1 = true;
		camEdge = false;
		break;
	default:
		break;
	}

	if ((engine->compositeLogEnabled) || (engine->toothLogEnabled)) {
		auto gap = getTimeNowUs() - lastEdgeTimestamp;
		addToothLogEntry(gap, trigger1, camEdge);
		lastEdgeTimestamp = getTimeNowUs();
	}
}

void LogCamTooth(trigger_value_e value DECLARE_ENGINE_PARAMETER_SUFFIX) {
	// bail if we aren't enabled
	if (!ToothLoggerEnabled) {
		return;
	}
	switch (value) {
	case TV_FALL:
		trigger2 = false;
		camEdge = true;
		break;
	case TV_RISE:
		trigger2 = true;
		camEdge = true;
		break;
	default:
		break;
	}

	if (engine->compositeLogEnabled) {
		lastEdgeTimestamp = getTimeNowUs();
		addToothLogEntry(lastEdgeTimestamp, trigger2, camEdge);
		}

}

void EnableToothLogger() {
	// Clear the buffer
	memset(buffer, 0, sizeof(buffer));
	memset(toothHistory, 0, sizeof(toothHistory));
	// Reset the last edge to now - this prevents the first edge logged from being bogus
	lastEdgeTimestamp = getTimeNowUs();
	thistoothtime = getTimeNowUs();
	nowUs = getTimeNowUs();
	// Reset write index
	NextIdx = 0;

	// Enable logging of edges as they come
	ToothLoggerEnabled = true;

	// Tell TS that we're ready for it to read out the log
	// nb: this is a lie, as we may not have written anything
	// yet.  However, we can let it continuously read out the buffer
	// as we update it, which looks pretty nice.
	tsOutputChannels.toothLogReady = true;
}

void EnableToothLoggerIfNotEnabled() {
	if (!ToothLoggerEnabled) {
		EnableToothLogger();
	}
}

void DisableToothLogger() {
	ToothLoggerEnabled = false;
	tsOutputChannels.toothLogReady = false;
}
ToothLoggerBuffer GetCompositeLoggerBuffer() {
	return { reinterpret_cast<uint8_t*>(buffer), sizeof(buffer) };

}
ToothLoggerBuffer GetToothLoggerBuffer() {
	return { reinterpret_cast<uint8_t*>(toothHistory), sizeof(toothHistory) };
}

#endif /* EFI_TOOTH_LOGGER */

```

shadowm60, 2020-12-02:
Can this be closed by #1902 ?

closed 2020-12-02 by rusefillc