rusefi/rusefi #1916: TCU input & output shaft RPM input

rusefillc, 2020-10-28:
Part of #1454

rusefillc, 2020-10-28:
@mck1117 proposes to use VSS for output shaft and probably primary RPM for input?

chuckwagoncomputing, 2020-10-28:
Supporting ISS and OSS sensors will definitely be a nice thing to have, but in addition we also need the ability to use VSS and RPM in their place. Basically this would mean having a TS field with a ratio so we can convert from VSS to OSS, as some vehicles use wheel speed sensors for VSS, while others use OSS for VSS.

This is a ways down on my to-do list, as we don't need to know ISS and OSS until we want to do closed-loop pressure control.

mck1117, 2020-10-28:
a while ago I was thinking of a VSS config that looked something like:
- teeth per revolution (at the sensor)
- sensor revolutions per driveshaft revolution
- sensor revolutions per wheel revolution
- wheel diameter/circumference/radius

That way we support all of the conceivable locations of the VSS: some cars have it on the transmission output, some have it on the axle, and some just use wheel speed sensors.

chuckwagoncomputing, 2022-04-21:
@mck1117 I propose we automatically use ISS for calculating the gear ratio if it is configured, and also provide RPM/ISS as "Torque Converter Ratio". Sound good?