rusefi/rusefi #1926: Smarter handing of very wrong firmware, F4<>F7 confusion

rusefillc, 2020-11-04:
Proteus F4 firmware on F7 hardware is alive enough to light up RED LED - but hangs up.

We should either hide F7 firmware way deeper or blink/handle F4 FW on F7 HW in an easier distinguishable way

mck1117, 2020-11-04:
Yeah, we should probably make the F7 firmware harder to find.  There isn't a reasonable way (in software, on the ECU) to detect that we're the wrong firmware, since the memory map is different enough that you're guaranteed to pretty quickly hard fault after startup.

mck1117, 2020-12-08:
In looking at random h7 stuff, there's nothing we can do here.  Major stuff is at completely different memory addresses, so if you load the wrong firmware, you can't do any IO without a hard fault

rusefillc, 2021-11-17:
we have improved console to confirm bundle type so it's solved on a different layer

closed 2021-11-17 by rusefillc