rusefi/rusefi #1927: Proteus TCU gear shift configuration

rusefillc, 2020-11-04:
I have two push buttons wires between GND and

```
	// "Digital 1" green
	engineConfiguration->tcuUpshiftButtonPin = GPIOC_6;
	// "Digital 6" white
	engineConfiguration->tcuDownshiftButtonPin = GPIOE_15;
```

actual results while powered only from +5v not +12v:

start-up
[image]

if I push button on WHITE "tcuDownshiftButtonPin " button I get "3" for a moment and then it returns to "4"

if I push button on GREEN tcuUpshiftButtonPin ONCE, I get _three_ back and forth 4 > 3 > 4 > 3 > 4 on the gauge

mck1117, 2020-11-04:
5v vs 12v shouldn't matter, the digital inputs on Proteus don't care how the board is powered.  So long as the switches are connected to ground, they should work fine.

rusefillc, 2020-11-04:
What debugging features do I have? Do we have button event counts / can we add button even counter?

rusefillc, 2020-11-04:
it looks like 0 and 1 are inverted which could be a configuration change but

@chuckwagoncomputing could "4" on start be a defect - do we expect button keep help keep ticking up or only tick once?

rusefillc, 2020-11-07:
we have three separate issues here:

1) in my configuration I needed to set "pullup" mode - that I will take care of

2) BUG: at the moment if you hold the button ON forever, "events" are happening with "timeout" frequency - that's a bug. there should be only 1 event if button is not being toggled

3) WAT? we have some weirdness in the code https://github.com/rusefi/rusefi/blob/7bfa18295a5e87883d3f56e6056f76cb03f74fa9/firmware/hw_layer/debounce.cpp#L102 maybe a merge conflict resolution has left stopConfiguration() method inside readPin where it totally looks unintended?

chuckwagoncomputing, 2020-11-07:
2 was actually intentional, but now that I think about it, it isn't what you would expect to happen.

mck1117, 2020-11-08:
wait, what is 1? the board has hardware pullups, the setting in the config does literally nothing (unless the input needs to be inverted?)

rusefillc, 2020-11-08:
yes, #1 does sw inverting

chuckwagoncomputing, 2021-01-02:
I'd like some more input on changing this (And whether we should). I implemented it as it is because some buttons you want to be able to hold to continuously increment, but now I see that perhaps that should be implemented on a higher level if it is desired.
Can we decide precisely what behavior we want?

rusefillc, 2021-01-02:
@chuckwagoncomputing let's begin from the beginning

what is the use-case for auto-increment? what is the user story here?

chuckwagoncomputing, 2021-01-02:
The first thing that comes to mind is cruise control speed adjustment, which I use all the time when I'm driving.
I've come to terms with the fact that auto-increment shouldn't be in the basic debounce method, even if it is somewhere in the debounce class.
My question is: What SHOULD be in the basic debounce method?

rusefillc, 2021-01-02:
since we have no cruise control plans (do we?) as of Jan 2, 2020 this is clearly a solution looking for a problem

my preference is to focus on very specific use-cases. I know that for starter we should NOT have forever starter in case of wiring issues. I know that in case of up-shift/down-shift i am NOT looking for auto-increment at this point.

chuckwagoncomputing, 2021-01-02:
1. If I hold the button, should the debounce function return true every time it is called? Currently, we have two functions, one that does and one that doesn't.

rusefillc, 2021-01-02:
the whole event polling model does not fit in my brain anyway

events mean listeners to me

my preference is not to touch anything here unless we HAVE to change things. please help me understand the urgency here.

chuckwagoncomputing, 2021-01-02:
So you think this issue should be closed?

closed 2021-01-02 by rusefillc