rusefi/rusefi #1941: config generator is very very slow due to heavy debug logging

mck1117, 2020-11-11:
`gen_config.sh` et al are quite slow, exclusively due to their use of logging every single thing that happens.

For example:

```
matthew@WIN-MOII5AJU4GS:/mnt/c/Users/matth/github/rusefi/firmware$ time ./gen_config_default.sh
... output snipped ...

real    0m5.901s
user    0m1.734s
sys     0m0.719s
matthew@WIN-MOII5AJU4GS:/mnt/c/Users/matth/github/rusefi/firmware$ time ./gen_config_default.sh > /dev/null
mkdir: cannot create directory ‘build’: File exists

real    0m0.561s
user    0m0.625s
sys     0m0.266s
```

10x speedup just by ignoring everything printed to stdout!

chuckwagoncomputing, 2021-01-04:
Is this on MacOS? I just timed it (on Linux) and it took 6.569 seconds.

rusefillc, 2021-01-04:
@chuckwagoncomputing this is not merged yet since I would like for helpful messages to not be just wiped out

mck1117, 2021-11-29:
gen is pretty quick these days

closed 2021-11-29 by mck1117