rusefi/rusefi #1943: occasional tunerstudio CRC errors

mck1117, 2020-11-12:
I see about 1 CRC error per second when TS is updating at ~100hz.

I'm pretty sure what's happening is that another thread is preempting the TS thread during the CRC calculation, and that other thread happens to write the output channel struct.  Then when the packet is sent, the CRC doesn't match the data!

rusefillc, 2020-11-15:
https://github.com/rusefi/rusefi/pull/1944 fails CI even with https://github.com/rusefi/rusefi/pull/1947

at the moment master is green with partial rollback - only "large" method is used https://github.com/rusefi/rusefi/commit/842372e69afd0657c98c17e121ebd41dec6315fd

rusefillc, 2020-11-15:
is it time to unit test both sr5WriteCrcPacketSmall and sr5WriteCrcPacketLarge?

rusefillc, 2020-11-16:
this rollback seems to be making the biggest difference - https://github.com/rusefi/rusefi/commit/7cba09b3faacd8b3ffa31aa928a47fae66157390 @mck1117

rusefillc, 2020-11-19:
👍

PR #1944 (2020-11-15): fix CRC errors
use the `crcReadBuffer` as a transmit buffer as well.  This ensures that we compute the CRC and transmit the same data - there's no opportunity for another thread to preempt and change the data between the CRC calculation and transmission.  This might actually net improve performance - it saves two calls to `sr5WriteData`.

fixes #1943
closed 2020-11-19 by rusefillc