rusefi/rusefi #1945: Boost control are useless with TPS as load axis

960, 2020-11-15:
Sorry to say, but boost control in open loop are useless as long as TPS are the load-axis.

The reason we use boost valves are to be able to lower the boost at some critical rpm-areas.

To get a better spool, and no creep, we need to keep the wastegate all shut until we have reached the desired boost.

With TPS as axis this isnt possible because the throttle are floored the whole pull(And in the top cell row all the time).

I searched up a easy explanation for this:

https://www.perrin.com/blog/post/boost-control-systems-explained-part-3

mck1117, 2020-11-16:
so then use PID, it'll keep the valve shut longer until you're closer to target.  you don't need much, just a little bit to keep the solenoid shut longer.

960, 2020-11-16:
I only use open loop for boost.

I've used a gp-pwm for it instead, so I'm good.

But would be nicer to use the function that are for it.
Tried changing the open-loop function to use 
map in firmware, added a load axis, but then it didnt work at all.

How a typical table I use looks.

[image]