rusefi/rusefi #1961: Fuel pump should turn on at the first sign of engine rotation

mck1117, 2020-11-17:
[image]

We were cranking for 0.7 second before we got valid RPM.  When we started injecting fuel, the pump hadn't caught up yet, so fuel pressure dropped almost 150kpa as the engine started.  Had the pump been turned on at the first sign of a trigger event occuring, the pump would already be up to speed by the time we started injecting fuel.  With compensation turned on I get a noticable improvement in starting quality since the pressure drop is compensated by additional pulse width.  That same benefit is available if we don't let the pressure drop in the first place.

rusefillc, 2020-11-18:
@mck1117 I will take care of it in the next 2-6 days

shadowm60, 2020-12-02:
Any updates on this topic? with the cold weather, faster start-up time is a good thing.

rusefillc, 2020-12-02:
this is falling down the priority list :( I will probably NOT make progress in the next two weeks :(

mck1117, 2020-12-03:
If we want to be a good, usable, smart ecu...

rusefillc, 2020-12-03:
If we want to be a good, usable, smart ecu in my opinion we have a lot of work in front of us.

@shadowm60 thank you for your DM with the magic code location, it was really helpful!

rusefillc, 2020-12-03:
@mck1117 @shadowm60 half of implementation is pushed
in my opinion we need to implement 'time_since_shaft_movement' or something like that

also see code comments wondering if Proteus has a bug which MRE does not have due to difference in EFI_MAIN_RELAY_CONTROL? it's a bit confusing let's put it this way :(

In my dreams we would need test coverage here as well :)

mck1117, 2020-12-03:
> EFI_MAIN_RELAY_CONTROL

eh? The main relay stays on fine, it's the fuel pump relay that doesn't turn on until valid RPM signal exists

mck1117, 2020-12-03:
the recently added `util/timer.h` class may be of use here

rusefillc, 2020-12-03:
@mck1117 @andreika-git  there is some magic which seems relevant here

```
	case LE_METHOD_TIME_SINCE_BOOT:
#if EFI_MAIN_RELAY_CONTROL
		// in main relay control mode, we return the number of seconds since the ignition is turned on
		// (or negative if the ignition key is switched off)
		return engine->getTimeIgnitionSeconds();
#else
		return getTimeNowSeconds();
#endif /* EFI_MAIN_RELAY_CONTROL */
```

mck1117, 2020-12-03:
Yes, the fuel pump primes for N seconds, then turns on once we get RPM.  In the time between getting RPM signal (we also start injecting fuel!) and the pump catching up, I lose almost HALF my fuel pressure.  Priming works fine.  The pump just needs to turn on earlier so the drop doesn't happen once injections start.

rusefillc, 2020-12-03:
fuel pump is usually controlled by FSIO
in my opinion we need to implement 'time_since_shaft_movement' for FSIO

see back-up non-FSIO implementation for reference

PR #2038 (2020-12-10): start fuel pump as soon as trigger events happen
fix #1961
closed 2020-12-10 by rusefillc