rusefi/rusefi #1969: Starter is engaged on start-up in pull-up configuration

rusefillc, 2020-11-19:
this issue is a bit different from #1965 which is already fixed:

now starter is ENGAGED on start-up until I touch toggle first time! timeout later starter is DISENGAGED, and at that point it starts to work as expected.

rusefillc, 2020-11-20:
"set engine_type 63" to set v12 config with start button and starter control
"debounce" to see input states
"gpio_pins" to see output states

rusefillc, 2020-11-26:
[image]

rusefillc, 2020-11-26:
[image]

rusefillc, 2020-11-26:
[image]

rusefillc, 2020-11-26:
https://github.com/rusefi/rusefi/commit/47fbb92bffab8a89a2ba9b4adb1acb63a7e369a5 has changed currentLogicValue default value from "#define INITIAL_PIN_STATE -1" since that currentLogicValue was breaking pull-up debounce use-case

rusefillc, 2020-11-27:
See https://github.com/rusefi/rusefi/pull/1990

mck1117, 2020-12-07:
is this fixed now?

closed 2020-12-07 by rusefillc