rusefi/rusefi #1987: build-in virtual dyno

rusefillc, 2020-11-25:
While we have https://github.com/rusefi/rusefi/issues/272 shall we build in some code to get a virtual dyno right in rusEFI?

we have VSS, we have RPM. do we need to calculate some acceleration?

https://github.com/BastianGschrey/PowerTune/issues/36 has some cool math

rusefillc, 2020-11-25:
[image]

rusefillc, 2020-12-14:
progress was made under #1999 

@shadowm60 @mck1117 what is the current status of this ticket? is this waiting for real vehicle testing?

shadowm60, 2020-12-14:
Exactly, wanted to check accel on vehicle. Next will be debug channel rename and finish hp calculation. Just fried my coils yesterday, need a few days to get new parts.

rusefillc, 2024-09-20:
@ElDominio do you remember this dream?

ElDominio, 2024-09-20:
2020 is before my times lol

rusefillc, 2025-05-07:
https://github.com/rusefi/rusefi/commit/a426bee9b8162913122cc62aad31f2face529194#diff-b8a7719a120bfcfc5ab1ba434549a0505a7ce79ea7c705e50ff5dc3ef016c618

rusefillc, 2025-05-07:
[image]

ggurov, 2025-05-07:
builtin virtual dyno like this is not very useful, at least not at the current stage when effort could be spent more productively elsewhere.

this functionality exists in MegaLogViewer already, in the form of calculated fields, they can use any gear, calculate from any ratio, using just RPM

[image]

[image]

however, the data seems too jerky to be visible nicely in the output, likely to do with high resolution of the data in the logs, so it comes out rather choppy and spiky:

[image]

perhaps a little bit of effort could be put towards attempting to make that work better via "smoothedRPM" field  of some sort that averages RPM for logging and using in MegaLogViewer

3eR01ive, 2025-05-23:
[image]