rusefi/rusefi #1996: Coolant temp shows "F" in new project

rusefillc, 2020-12-02:
I am not getting it. How is my coolant gauge still showing "F" even in brand new project?

[image]

rusefillc, 2020-12-02:
@mck1117 it's a it's a bug/feature????!!!! in TS
if you re-pick the gauge (right click menu) it shows C

rusefillc, 2020-12-02:
One day someone should ask TS people.

mck1117, 2020-12-02:
email sent to TS folks

mck1117, 2020-12-02:
`#set CELSIUS`

and/or

`useLegacyFTempUnits = false`

closed 2020-12-02 by rusefillc

rusefillc, 2022-01-08:
@ElDominio it says "F" while in reality it's "C"

it's TS bug which TS has resolved in recent beta, not yet resolved in release of TS

workaround is to change gauge to something not temperature and then to switch it back to CLT

PR #1998 (2020-12-02): disable legacy temp units
fix #1996