rusefi/rusefi #2001: all triggers should be documented way nicer

rusefillc, 2020-12-03:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers is lame

we can do better

rusefillc, 2020-12-03:
@ksmola I wanted to ask your help with design of individual images and overall https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers

[image]

rusefillc, 2020-12-03:
[image]

rusefillc, 2020-12-05:
See also https://github.com/rusefi/rusefi/issues/2022

mck1117, 2020-12-06:
I believe this to be the tdc mark

[image]

based on

[image]

rusEFI trigger sync point is at 310 degrees ATDC #1 (aka trigger offset = 50)

rusefillc, 2020-12-09:
by the way we can learn from https://gitlab.com/libreems-suite/ardu-stim/-/blob/master/ardustim/wheel_defs.h

rusefillc, 2020-12-10:
just for reference, before flip & TDC on top

[image]

[image]

rusefillc, 2020-12-11:
[image]

rusefillc, 2020-12-11:
[image]

rusefillc, 2020-12-11:
[image]

rusefillc, 2020-12-11:
subaru

[image]

[image]

rusefillc, 2020-12-11:
[image]

rusefillc, 2020-12-12:
@chuckwagoncomputing this depends on #2059 2059

rusefillc, 2020-12-14:
A lot of progress was made here, closing this one - will create separate tickets if new ideas (and time) happen

closed 2020-12-14 by rusefillc