rusefi/rusefi #2006: MRE pin reuse state validation fails

rusefillc, 2020-12-03:
[image]

rusefillc, 2020-12-04:
See https://rusefi.com/forum/viewtopic.php?p=38731#p38731 for more info

it looks like we still have TLE8888 feedback and that feedback is what's causing the issue with pin state validation?

rusefillc, 2020-12-04:
@dron0gus

mck1117, 2020-12-10:
> it looks like we still have TLE8888 feedback and that feedback is what's causing the issue with pin state validation?

I don't think that's the problem.  In this case the pin was set to inverted, so we expected to read back a 1.  Which...we did?  So it's clearly not a hardware problem.  Software is interpreting the pin read wrong.

mck1117, 2020-12-10:
Nope.  It's better than that.  The bug is that the validation check didn't work at all with pins set to inverted.

PR #2051 (2020-12-11): reduce GPIO complexity & fix #2006
- unconditionally set physical output pin every time.  Checking whether we need to costs about the same as actually setting the pin (which you'd have to do anyway), so just set the pin and don't bother checking
- re-enable pin state validation

Tested with a logic analyzer that GPIO does indeed still work correctly.

fixes #2006
closed 2020-12-11 by rusefillc