rusefi/rusefi #2010: 6.1 Hemi V8 decoder

rusefillc, 2020-12-04:
https://speeduino.com/forum/viewtopic.php?f=12&t=80&start=160#p42604

[image]

"That 36+2-2 just like srt4 neon"

rusefillc, 2020-12-04:
Well, that's a tricky wheel! it's symmetrical  for one front (16 + 16 same events) and not symmetrical on the other front (15 + 17)

weird thing is that for this to be decoded by rusEFI we would need trackingGap = 17!

Current implementation of "trackingGap = 17" would check all 17 gaps deep while this wheel decoder would be more efficient if only gap #0 and gap #14 is checked

rusefillc, 2020-12-04:
in rusEFI it's oriented like this

[image]

rusefillc, 2020-12-04:
from right-middle point (rusEFI wheels rendering start at right-middle point), counter clockwise:

16 teeth
huge gap
15 teeth
huge tooth

===================

current reality:

16 teeth
huge gap
15 teeth
huge tooth

it matches!

[image]

closed 2020-12-04 by rusefillc

mck1117, 2021-03-07:
wait - does this actually need huge gap tracking length? the huge tooth and huge gap are different widths - one is missing 2 teeth, and the other is maybe 3

commit 6200569f (2020-12-04): 6.1 Hemi V8 decoder fix #2010