rusefi/rusefi #2011: ADC Issue after adc cleanup/change a while ago

960, 2020-12-04:
After the ADC files was changed a while back, the adc issues returned.

Behaving just like ADC issues I had long time ago.

I use F7

Did a screen recording how it looks like.
This is by applying 0 volts and 5 volts, and you can see it moves in steps and hangs:

If I apply 5 volts again before it falls down to 0, it uses the short range where it hangs and up.

Same if I let it fall to 0, it will use the range from 0 to the first step it hangs.

If I use the older ADC files everything are ok, 

[PrKYOq5x2a.zip](https://github.com/rusefi/rusefi/files/5643400/PrKYOq5x2a.zip)

960, 2020-12-04:
Another difference is with old files Mcu temperature shows -25 degrees

With new files it shows right

rusefillc, 2020-12-04:
@960 do you use 100 pin or 144 pin package? do you have an option to fabricate your board with F4?

960, 2020-12-04:
I have 144 pin package.

STM32F767ZIT6

JLC had F7 mcus now :-)

I have F7 at all my boards, not sure if I will fabricate more of them yet.
But it will be messy to have two different MCU's

It's the exact same issue as before with the cache-thing Matt fixed.

Just been using the older files.
Not sure what breaks it, as it just looks like a cleanup..

mck1117, 2020-12-05:
> @960 do you use 100 pin or 144 pin package? do you have an option to fabricate your board with F4?

package doesn't matter, they're the same die in a different piece of plastic

dron0gus, 2020-12-05:
@960 
What non-standard peripheral is used on your board?
May be SPI5, UART6, SPI4? All of them are using same DMA channels as ADC1,2,3.

dron0gus, 2020-12-05:
See https://github.com/dron0gus/rusefi/commit/ade5d3a767c6a9f03b86e7fa3203fd78e2e374d3

960, 2020-12-05:
I am only using SPI3 for FRAM, and SPI2 for CJ125.
Rest is just normal GPIO.

CJ125 are disabled as I have external wideband, and FRAM are only active when I hit burn.

dron0gus, 2020-12-05:
@960 can you try https://github.com/rusefi/rusefi/pull/2021 . Seems my bug.

960, 2020-12-05:
@dron0gus Now it seems to work fine.

Thanks!

rusefillc, 2020-12-05:
@mck1117 package matters for possibility of F4 swap

closed 2020-12-05 by rusefillc